mantidproject/mantid #34586: Remove redundant Code for Old OS/Package Management Support

martyngigg, 2022-10-20:
**Describe the outcome that is desired.**
Cleaner CMake without baggage of support for RHEL7/Ubuntu 18.04 now they have been dropped

**Describe any solutions you are considering**

At a minimum this should cover:
- [ ] Remove Windows ThirdParty CMake links
- [ ] Remove RHEL7 packaging rules
- [ ] Check with ILL if they still use the debian packaging in the Singularity images
- [ ] Remove pre-conda build scripts for Jenkins

sf1919, 2024-04-16:
@quantumsteve and @peterfpeterson - how far does #37156 resolve this issue?

peterfpeterson, 2024-04-16:
#37156 is actually part of the cleanup that wasn't mentioned on the list. It is towards the same goal though.

thomashampson, 2024-08-08:
I think this is already all cleaned up

closed 2024-08-08 by thomashampson