mantidproject/mantid #34606: Show ROIs as ranges rather than lists

gemmaguest, 2022-10-24:
On the ISIS Reflectometry interface Preview tab, when you apply the selected instrument viewer ROI to the Experiment settings, each individual selected detector gets listed in the table, creating a very long string. We should consolidate contiguous ranges of detectors into a grouping pattern string instead, e.g. `1-5, 10-12` instead of `1,2,3,4,5,10,11,12`. There may be some string utilities in Kernel or elsewhere we could use to help with this.

ConorMFinn, 2022-11-04:
This appears to have been done by another PR, though I'm struggling to find which one it is. Any idea @gemmaguest?

gemmaguest, 2022-11-04:
@ConorMFinn It works for the ROIs from sliceviewer but not for the detector IDs from instrument viewer, I think?

ConorMFinn, 2022-11-04:
Ah! I see, forgot that section got put in as well. Thanks!

PR #34681 (2022-11-29): Show Detector ROIs as Ranges on the Experiment Settings Tab
**Description of work.**
Make the Apply button on the preview tab fill the Experiment Settings tab's ROI Detector IDs column with ranges rather than an unreadable list of values. 

**To test:**
1. Open the ISIS Reflectometry interface
2. Go to the Preview tab
3. Enter run `INTER45455_inst` and click Load (This data is available on this Epic and must be in your path: #31069)
4. Select the `Draw a Rectangle` button on the instrument viewer. 
5. Draw over some of the detectors. 
6. Hit Apply
7. Check that the selected detectors are now p [... cut]
closed 2022-11-29 by martyngigg