mantidproject/mantid #34633: Add option for negative reflection to indirect Symmetrize

Harrietbrown, 2022-10-28:
In the indirect Symmetrize tab we use the Symmetrize alg to reflect the positive range onto the negative, it has been requested that we implement a method so that instead we can reflect the negative onto the positive. 

A method to implement this is to run ScaleX multiplying by -1 and then SortXscale,

AnthonyLim23, 2022-11-29:
[image] [Label for option in sample changer](https://trello.com/c/hdN0QZQY/16-label-for-option-in-sample-changer)

AnthonyLim23, 2022-11-29:
[image] [Negative option for indirect symetrize](https://trello.com/c/1z6SEVVv/26-negative-option-for-indirect-symetrize)

PR #34932 (2023-01-30): Add negative reflection option to indirect data manipulation symmetrise tab
**Description of work.**
This PR adds an option for negative reflection in the indirect data manipulation symmetrize tab. Previously, the user can only reflect the positive range of the x values to the negative range but now they can reflect the negative range of the x values to the positive.

**To test:**

Go to Interfaces->Inelastic->Data Manipulation->symmetrise:
1- Load some data 
2- Try both the "Positive to Negative" and "Negative to Positive" options
3- Make sure when you change the option the values of the EMin [... cut]
closed 2023-01-30 by DannyHindson