mantidproject/mantid #34637: Dockable widgets in Indirect Data analysis can not be put back vertically

Harrietbrown, 2022-10-28:
In the Indirecty Data Analysis fit tabs we utalise dockable widgets, there should be able to be put back where they were removed from. 
however if you remove the data selector (normaly stored above the other widgets) and try to place it back it will only go back next to the other widgets horizontally. this issue is to find a solution that will always allow for widgets to be replaced vertically.

AnthonyLim23, 2022-11-25:
could this be an induction issue?

Harrietbrown, 2022-11-25:
It could well be, so long as they have had the qt tutorial

AnthonyLim23, 2022-11-29:
[image] [Dockable widgets cannot be put back vertically](https://trello.com/c/hGxLoOPz/25-dockable-widgets-cannot-be-put-back-vertically)

PR #35083 (2023-01-27): Fix data selector can not be put back vertically in indirect data analysis tabs
**Description of work.**
This PR fixes the indirect data analysis data selector widget. Previously, the tab cannot be returned to its original position.

**To test:**
1- Open Scientific Interfaces->Indirect->Data Analysis
2- Drag the data selector widget outside the window
3- Put back the widget into the window

Fixes #34637. 

*This does not require release notes* because **it is a small UI improvement**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.or [... cut]
closed 2023-01-27 by martyngigg