mantidproject/mantid #34658: Bayes Stretch

AnthonyLim23, 2022-11-01:
The Bayes stretch algorithm https://docs.mantidproject.org/nightly/algorithms/BayesStretch-v1.html uses the Quest module from quasieleasticbayes, to output a contour plot of the "goodness of fit" as a function of sigma and beta. This is achieved by fixing the tau and beta values and evaluating fits along a grid.

With the new Python implementation this could be very slow and it is not used much in Mantid (see below). Is it worth us continuing to support this algorithm?

name | mantidVersion | sum
-- | -- | --
BayesStretch.v1 | 3.10 | 2
BayesStretch.v1 | 3.11 | 5
BayesStretch.v1 | 3.12 | 10
BayesStretch.v1 | 3.13 | 23
BayesStretch.v1 | 3.5 | 4
BayesStretch.v1 | 3.6 | 1
BayesStretch.v1 | 3.7 | 3
BayesStretch.v1 | 3.8 | 2
BayesStretch.v1 | 3.9 | 3
BayesStretch.v1 | 4.0 | 4
BayesStretch.v1 | 4.1 | 7
BayesStretch.v1 | 4.2 | 5
BayesStretch.v1 | 5.0 | 1
BayesStretch.v1 | 5.1 | 8
BayesStretch.v1 | 6.0 | 5
BayesStretch.v1 | 6.1 | 3
BayesStretch.v1 | 6.2 | 11
BayesStretch.v1 | 6.3 | 3
BayesStretch.v1 | 6.5 | 1

AnthonyLim23, 2022-11-01:
@martyngigg and @SpencerHowells what do you think?

martyngigg, 2022-11-07:
@AnthonyLim23 Thanks for stats on this. How slow are we talking? If necessary would there be opportunities for speedup with things like `numba` if required in the future?

With those numbers though it's hard to justify much work on it.

AnthonyLim23, 2022-11-07:
Very very slow. However, it was because I was doing an inefficient search. I will have another look at it once I am happy QSe (the hope is to just stick QSe inside a loop)

closed 2023-04-06 by AnthonyLim23