mantidproject/mantid #34659: Remove ResNorm

AnthonyLim23, 2022-11-01:
The ResNorm algorithm uses the quasielasticbayes package. However, the module contains "licence software", so we should not use it. The ResNorm algorithm does have a version 2, and version 1 is unused. So this issue is to remove it 

https://github.com/mantidproject/mantid/blob/dfa6d374579bbae1fecabd9d469a0c182d14db70/Framework/PythonInterface/plugins/algorithms/ResNorm.py#L96

AnthonyLim23, 2022-11-29:
[image] [Remove ResNorm](https://trello.com/c/w9cecp02/19-remove-resnorm)

MohamedAlmaki, 2022-12-23:
I've removed the algorithm python file and its directory from CMakeLists.txt. Do I need to remove the IndirectImport and quasielasticbayes also?

AnthonyLim23, 2023-01-04:
Those files are used elsewhere. You can check by using grep to see where the imports are used.

PR #34903 (2023-01-10): Remove ResNorm V1 algorithm
**Description of work.**
This PR removes unused ResNorm V1 because it uses license software.

**To test:**

Check that the first version of the algorithm doesn't exist in Algorithms->Workflow->MIDAS 

Fixes #34659. 

*Not sure, where to add the release note*

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are  [... cut]
closed 2023-01-10 by martyngigg