mantidproject/mantid #34692: Jenkins Build Artefacts Are Unclear

ConorMFinn, 2022-11-09:
**Describe the bug**
The builds artefacts (which I believe are related to the unreliable test tracking after a discussion with @sf1919) are misleading. If there is a build error, the [build artefact appears to show all tests passing](https://builds.mantidproject.org/job/pull_requests-conda-osx/520/) and given its name it appears to be the outcome of the build (`darwin17.log`, `linux-gnu.log`, etc.). The actual build errors are only visible in the console output. 

If these output logs could be renamed to make it clear what they're for it would reduce a lot of the confusion. Including a clear comment at the top/bottom of the log file may also work well. 

**To Reproduce**
1. Create a PR with some build errors. 
2. Check the logs on the jenkins page. 

**Expected behavior**
The artefacts should be clearly named so it is obvious that they should not be used for checking the outcome of your build. 

**Platform/Version (please complete the following information):**
 - OS: All

DonaldChung-HK, 2022-12-09:
System tests and unit tests are executed independently in the pipeline. The log files only contain information for unit tests executed by `ctest`. 

There are 2 solutions for this issue:
1. Renaming the log files to show that they are for unit tests only (However, this might cause disruption on the unit test monitoring system)
2. Integrate system tests into ctest so that both can be monitored by the dashboard.

thomashampson, 2023-11-14:
This was fixed. `ctest` was prepended to the file name.

closed 2023-11-14 by thomashampson