mantidproject/mantid #34716: Add fields to support sparse workspace features in Indirect, Corrections UI

DannyHindson, 2022-11-11:
The "Calculate Monte Carlo Absorption" tab in the Indirect, Corrections UI calls the PaalmanPingsMonteCarloAbsorption algorithm to perform an absorption correction. This algorithm supports a sparse workspace feature which calculates the correction on a subset of the detectors on an instrument and then uses interpolation to return a result for the remaining detectors. This speeds up the calculation on instruments with a lot of detectors.

This change is to add the sparse workspace fields to the UI so the users can take advantage of this feature. The following screenshot of the algorithm dialog for PaalmanPingsMonteCarloAbsorption shows the required fields:

[image]

Requested by Spencer Howells

AnthonyLim23, 2022-11-11:
this is a harder induction issue

AnthonyLim23, 2022-11-29:
[image] [Add fields to support workspace features in indirect corrections GUI](https://trello.com/c/slQulEzY/23-add-fields-to-support-workspace-features-in-indirect-corrections-gui)

PR #34882 (2023-01-13): Add sparse workspace features support to the indirect correction UI
**Description of work.**
This PR adds sparse workspace features support to the "Calculate Monte Carlo Absorption" tab in the Indirect Corrections UI. Three fields have been added to the UI spare instrument checkbox, the number of detector rows and the number of detector columns. Note, that these fields are already present in the PaalmanPingsMonteCarloAbsorption algorithm UI, so I just added them to the correction UI.

**Report to:** Spencer Howells

**To test:**

From the top bar open Interfaces->Indirect->Corrections. Then op [... cut]
closed 2023-01-13 by martyngigg