mantidproject/mantid #34739: Mantid Help crashes in Windows Conda install

thomashampson, 2022-11-16:
**Describe the bug**
Attempting to open the help on Windows will crash for workbench that has been installed with Conda, with the following error message in the terminal:
`Could not find QtWebEngineProcess.exe`
The help interface in the standalone package is fine.

**To Reproduce**
1. Install mantid from Conda:
```
conda create --name mantid-test
conda activate mantid-test
conda install -c mantid mantidworkbench
```
and launch workbench with
```
mantidworkbench
```
2. Try to open the help from the menu bar: `Help->Mantid Help`

**Expected behavior**
Help should display

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: Windows 10
 - Mantid Version: 6.4 onwards mantidworkbench Conda package

**Additional context**
This has been an issue since the start of our Windows Conda mantidworkbench package, but only since 6.5 has the error message appeared.

jhaigh0, 2023-01-05:
in `mantidworkbench.cpp` there is a comment which says

```
It was observed on Qt >= 5.12 that the QtWebEngineProcess would fail to
load the icudtl.dat resources due to Chromium sandboxing restrictions. It
would appear there is no more fine-grained way to control the restrictions:
https://doc.qt.io/qt-5/qtwebengine-platform-notes.html
```

jhaigh0, 2023-01-05:
https://stackoverflow.com/questions/67060913/pyinstaller-could-not-find-qtwebengineprocess-exe-for-pyqt5-web-application

This post recommends putting
```
[Paths]
Prefix = PyQt5/Qt
```
in the `qt.conf` file to fix the problem.

Currently, in `installers/conda/common/qt.conf` (which I think is the one to look at) it says
```
[Paths]
;; This assumes this files is beside an executable in a directory one-level deep
;; from the root of an installed conda environment.
Prefix = ./../
```

jhaigh0, 2023-01-09:
This page has good information
https://doc.qt.io/qt-5/qtwebengine-deploying.html

"
At runtime, Qt WebEngine looks for the QtWebEngineProcess executable in the directory that [QLibraryInfo::location](https://doc.qt.io/qt-5/qlibraryinfo.html#location)([QLibraryInfo::LibraryExecutablesPath](https://doc.qt.io/qt-5/qlibraryinfo.html#LibraryLocation-enum)) returns. For Qt installations, this is QTDIR/libexec (Linux) or QTDIR\bin (Windows). The path can be changed by defining a `qt.conf` file, for example. Alternatively, an executable path can be set as a value of the QTWEBENGINEPROCESS_PATH environment variable. On macOS, Qt WebEngine looks for the executable in `.app/Helpers/QtWebEngineProcess`.
"

thomashampson, 2023-07-18:
this is still something we should attempt to resolve

adriazalvarez, 2024-02-21:
Crash still found in v6.9 smoke testing with Conda on Windows.

thomashampson, 2024-03-05:
We discussed the idea of removing the Qt web engine for Windows and ship only the html files, then have help open in a browser.

thomashampson, 2025-05-28:
for now leaving this in the 6.13 milestone as it may be resolved in this release

closed 2025-09-27 by mantid-builder