mantidproject/mantid #34740: Plotting 3D wireframe causes out of range error

robertapplin, 2022-11-16:
**Original reporter:**
Pascal Manuel

**Describe the bug**
Selecting a number of workspaces and attempting to plot a 3D wireframe causes an error in Workbench. This however used to work in MantidPlot.

**To Reproduce**
1. Load the data in the following zip files

[53022-2_9raw-d.zip](https://github.com/mantidproject/mantid/files/10022062/53022-2_9raw-d.zip)
[53023-2_9raw-d.zip](https://github.com/mantidproject/mantid/files/10022064/53023-2_9raw-d.zip)
[53024-2_9raw-d.zip](https://github.com/mantidproject/mantid/files/10022066/53024-2_9raw-d.zip)

3. Select all the workspaces in the ADS
4. Right click, and choose `Plot`->`3D`->`Wireframe`
5. A sad mantid appears and you get this error:

```
Traceback (most recent call last):
  File "C:\MantidInstall\bin\lib\site-packages\workbench\plugins\workspacewidget.py", line 228, in _do_plot_3D
    plot_function(workspaces)
  File "C:\MantidInstall\bin\lib\site-packages\mantid\plots\plotfunctions.py", line 61, in inner_func
    return func(workspaces, *args, **kwargs)
  File "C:\MantidInstall\bin\lib\site-packages\mantidqt\plotting\functions.py", line 367, in plot_wireframe
    fig.clf()
  File "C:\MantidInstall\bin\lib\site-packages\matplotlib\figure.py", line 988, in clf
    return self.clear(keep_observers=keep_observers)
  File "C:\MantidInstall\bin\lib\site-packages\matplotlib\figure.py", line 2814, in clear
    super().clear(keep_observers=keep_observers)
  File "C:\MantidInstall\bin\lib\site-packages\matplotlib\figure.py", line 956, in clear
    ax.cla()
  File "C:\MantidInstall\bin\lib\site-packages\mpl_toolkits\mplot3d\axes3d.py", line 1143, in cla
    self.set_zlim(0, 1)
  File "C:\MantidInstall\bin\lib\site-packages\matplotlib\cbook\__init__.py", line 1812, in method
    return getattr(self, name)(*args, **kwargs)
  File "C:\MantidInstall\bin\lib\site-packages\mantid\plots\mantidaxes.py", line 1340, in set_zlim3d
    self._set_overflowing_data_to_nan(2)
  File "C:\MantidInstall\bin\lib\site-packages\mantid\plots\mantidaxes.py", line 1386, in _set_overflowing_data_to_nan
    self.collections[0].set_segments(all_data)
  File "C:\MantidInstall\bin\lib\site-packages\matplotlib\axes\_base.py", line 1373, in __getitem__
    return [artist
IndexError: list index out of range
```

**Expected behavior**
A 3D wireframe plot should appear.

**Platform/Version (please complete the following information):**
Windows, IDAaaS, and probably all others. Mantid v6.5

robertapplin, 2023-01-04:
That would be great if thats the case! @jhaigh0 If you `conda install matplotlib=3.5.*` within your environment does the bug reappear?

jhaigh0, 2023-01-04:
Yes, in that case I can recreate the error 👍

robertapplin, 2023-01-04:
Thanks for checking, in that case we can close this! 😄

closed 2023-01-04 by robertapplin