mantidproject/mantid #34744: Specify polarisation efficiencies via a workspace

gemmaguest, 2022-11-16:
Currently in the ISIS Reflectometry interface, polarisation analysis can be done but is limited to using efficiencies specified in the Instrument Parameter File. For POLREF, users need more flexibility and would like to be able to provide a workspace/filename instead. We should add an option similar to the flood correction to be able to use a workspace or the parameters file. (Although check whether the parameters file method is still useful; if not, remove it.)

[image]

It would be useful to do #34720 first and then we can use the same pattern for the polarisation corrections.

**Implementation**
I think it should be straightforward to optionally use a passed-in workspace instead of one extracted from the parameters file.
- In `ReflectometryReductionOneAuto` add input properties to pass the workspace and correction method (`Efficiencies` and `CorrectionMethod` from `PolarizatioEfficiencyCor`).
- `ReflectometryReductionOneAuto::getPolarizationEfficiences()` currently creates the efficiencies workspace from the parameters file. It should use the passed-in workspace instead, if one is given. Note that this method also returns a "correction option" which indicates the type of polarisation analysis being done; this can be inferred from the number of workspaces in the input workspace group (2=>PNR, 4=>PA).
- `ReflectometryISISLoadAndProcess` should take a filename and do the necessary loading then pass it to RROA as a workspace. It should also expose the `CorrectionMethod` property.

rbauststfc, 2022-11-24:
Notes from requirements meeting with Andrew:

This work would be ideal to have in a nightly for an experiment taking place on the 13th of December (for ease of reducing a large volume of data), but if we find we can't meet that deadline then we need to let Andrew know by 9th December so he can revert to contingency plan.

Clarification on the questions above is that currently the corrections are being applied via a Jupyter notebook, so it makes sense that there are no parameter file settings for POLREF corrections. Usually they would use a Wildes correction, but they require quite a bit of flexibility so hard-coding into the parameter file is really not ideal for them. It would be great if we could infer the correction method from the efficiencies workspace passed in, but it's also fine for them to have to specify the method to use if that proves easier.

For inferring the flipper configuration for a Wildes correction, this can be figured out from the number of workspaces in the input workspace group (same as Fredrikze). If the number of workspaces in the group is 2 then flippers should be `0, 1`, if the number of workspaces is 4 then flippers should be `00, 01, 10, 11`. Technically a larger range of flipper configurations is possible, but not used currently on POLREF. Certainly for the purposes of the December experiment we should be able to use this logic and could look at making it more flexible over time.

rbauststfc, 2022-11-24:
Update on this - I've pushed branch `34744_specify_polarisation_efficiencies` which has the following progress:

- I have added a new optional parameter to `ReflectometryReductionOneAuto` called `PolarizationEfficiencies` that can be used to pass in an efficiency workspace. This is the only new parameter we should need as the rest can be worked out.
- I have updated the logic in `ReflectometryReductionOneAuto::getPolarizationEfficiences()` so that when an efficiency workspace is provided it will use that to find the correction method. I have also added the logic to infer the correction option for the relevant correction method. Some of this would be good to improve when we have time, but I think it would do the job for the December experiment.
- Trying to set up a unit test is a bit complicated, so I did some temporary hacking to check the changes I've made using the existing tests. Since then I've been looking at trying to set up a proper unit test. I have the beginnings of this which I will commit to the branch at the end of today, but it's not currently working properly and should be considered WIP - it can be discarded if not useful.

The remaining steps on the algorithm side are to complete the final two steps listed in the main issue description. Note that there will only be the one new `PolarizationEfficiencies` parameter to add to `ReflectometryISISLoadAndProcess` and `ReflectometryReductionOneLiveData` as it wasn't necessary to add a `CorrectionMethod` parameter in the end.

I haven't made a start on the GUI side of this yet. Depending on preferences, whoever is picking this up could continue working on the algorithm changes or could consider making a start on the GUI side. The GUI work should be largely independent of the algorithm changes up until the point where we need to pass data to the new algorithm property, so it may be a good way of speeding up the work. Once the algorithm work went into PR we could then pair programme connecting up the two sides.

PR #34809 (2022-12-05): Add Polarization Efficiencies to GUI
# BLOCKED BY: 
- [ ] #34805

**Description of work.**
Adds a GUI element to allow selection of a workspace for the new Polarization Efficiencies property of the Refl Load and Process algorithm. 

The logic is set up such that the check box enables and disables the polarization correction, then the combo box allows selection of workspaces or to be left blank should the user wish to use the values in the parameter file. 

**To test:**
1. Talk to Conor or Rachel about getting the corrections file. It's not public, so it can't be put here. 
2. Load the file int [... cut]