mantidproject/mantid #34760: Bilby InstrumentView Side by Side view improvement

robertapplin, 2022-11-18:
**Original reporter**
Liliana De Campo

**Description**
A request to improve the Side by side view on InstrumentViewer has been made by a Bilby instrument user. A similar request was made by SXD users [here](https://github.com/mantidproject/mantid/issues/34560), but I thought I'd record this in a different issue. It might be worth running changes to the InstrumentViewer Side by side view past Liliana De Campo to see if her issues are also solved by this work.

The following was provided in their email:
```
When looking at 2D Bilby data, the 3D view is perfect!

But in the side-to-side view, the sequence of the detector tubes is messed up:
the tubes are organised in 8-packs, and the picture shows that 8 peaks with high intensity are at the wrong place; red circles
also the center of the back detector (direct beam area) lands at a funny position; orange circles
and they are not all aligned (overall shape of the detector panels is not a smooth rectangle), see blue circles)

It seems that this side-by-side view is a little different in different Mantid versions, but none of them is perfect.

A better side-so-side viewer would make it much easier for us to mask different detector areas
```

[image]

DannyHindson, 2023-02-01:
I've had a look into these problems with the Bilby instrument display in side by side view. The main issue is that the code is identifying the banks at the wrong level. The code identifies 30 banks rather than 6. This happens due to the two level structure in the Bilby instrument definition where each bank consists of 5 "eight packs":
[image]

I am going to have a look at whether the logic in PanelsSurface::processTubes can be modified to identify 6 banks in this case. There's already some logic to handle the tubes being separated from the bank by intermediate levels (eg tubes being grandchildren of the bank) but it requires the intermediate levels to have a single child. In the Bilby case we have one intermediate level where the components in this level have 8 children

If I hack together a solution for this to see what it looks like I get this:
[image]
The 6 banks are now clearly visible. The arrangement of the banks however is a bit "random". This is due to the spread banks logic in the side by side view. After projecting the banks onto the xy plane the banks are then moved away from each other to remove any overlaps. If a different arrangement would be preferable then the change in the linked PR for SXD may be useful where the bank centre positions can be specified

DannyHindson, 2023-02-01:
I will send Liliana an email about this

DannyHindson, 2023-02-01:
Email exchange with Liliana at ANSTO:

`Hello Danny`

`Thank you so much for looking at this and explaining to me!`

`Your hacked side-by-side view looks very promising to me, although I believe that some of the images may be mirror reversed? And it would be great if the rear detector panels would join together perfectly to one piece in the center of the view (as in the 3D view).`

`Here is a link to a BBY file that I think would make this easier to see:`

`https://cloudstor.aarnet.edu.au/plus/s/j7dBLMlaKd9QwGS`

`And here is a sketch for the comparison between 3D view and what I believe would be the ideal side-by-side view:`

[image]

`Many thanks!`
`lela`

`From: Danny Hindson - STFC UKRI <[danny.hindson@stfc.ac.uk](mailto:danny.hindson@stfc.ac.uk)> `
`Sent: Wednesday, 1 February 2023 9:23 PM`
`To: DE CAMPO, Liliana <[lilianad@ansto.gov.au](mailto:lilianad@ansto.gov.au)>`
`Subject: Mantid side by side view for Bilby instrument`

`Hi Liliana,`

`I’ve been doing some work on the instrument side by side view for another ISIS instrument recently and I wanted to get in touch about this problem you reported a few months ago because there’s a bit of overlap.`

`So I think I’ve found what the problem is for the Bilby instrument. The way the side by side view works is that it does the following:`
`a)	Identifies flat banks of tubes from the instrument definition`
`b)	Rotates the banks so they are parallel to the xy plane (where z=beam)`
`c)	If any of the banks overlap it spreads them out – away from the largest bank`

`For the Bilby instrument step a) is going wrong. The code is identifying banks at the wrong level in the instrument definition. It is looking at the eight pack level and it decides there 30 banks instead of 6. When actions b) and c) then proceed the eight packs from a single bank get separated and moved into weird positions. I can probably fix this by making a code change to a). I’ve quickly hacked together a solution so you can see what this would look like:`

[image]

`You can now see the 6 banks. Although the bank locations are still perhaps a bit “random”. The change I’ve done for the other ISIS instrument is to allow the bank positions in the side by side view to be specified in the instrument definition file. So if the above screenshot looks OK apart from the overall bank locations then I can probably sort this out for you. Please can you let me know if this looks good? If yes then feel free to send me a sketch the bank arrangement you’d like in the side by side view and I could send you a further screenshot of how this would look`

`Thanks`
`Danny`