mantidproject/mantid #34769: Implement SXD analytical absorption correction algorithm

DannyHindson, 2022-11-22:
Implement an analytical absorption correction method in Mantid for use on the SXD instrument. As discussed with Silvia Capelli. The method decomposes a mesh sample shape into a set of tetrahedra and then uses an analytical solution for the absorption effect for each tetrahedron to calculate an overall absorption correction for the entire shape. The method is described in this paper:

[acta_19_1965_Tompa.pdf](https://github.com/mantidproject/mantid/files/10066319/acta_19_1965_Tompa.pdf)

There is already an implementation of the algorithm in C that Alberto Leonardi has written. It's not clear yet to what extent that code can be ported into Mantid vs doing a new implementation. As a minimum that C implementation will be a useful benchmark and point of reference while doing the Mantid work on this.

DannyHindson, 2022-12-08:
Brief summary of method with some references to Alberto's code (unit _Utility_CrystalAbsorption.cpp_):

Repeat following calculation for each outgoing beam direction:
1) Decompose crystal shape into Howell's polyhedra (_CrystalAbsorption_SXD_Mesh_Get_SubHowellSet_)
2) Decompose each polyhedron into a set of tetrahedra (_CrystalAbsorption_SXD_Mesh_Get_SubHowellSet_Tetarhedron_)
3) Calculate absorption due to each tetrahedron and sum (_CrystalAbsorption_SXD_Mesh_Get_TransmissionContributions_)

The method outlined in the Tompa paper has some corrections to the maths in the following places in Alberto's code. I've looked through the paper and the corrections look correct to me:

L3183, minus sign missing in the paper is inserted
L3176/L3180, factor of minus applied. Definition of -h'(u) in paper is out by a factor of -1

DannyHindson, 2022-12-08:
Estimate for this work. If I do the work then some of the items might be less because I've digested the method now but best to go with these numbers:

| Task          | Estimate           |
| ------------- |:-------------:|
| Review Tompa paper and understand algorithm | 5 |
| Port\write code                                                    | 5 |
| Write tests | 2 |
| Docs | 1 |
| PR process and review | 1 |
| **Total** | **14 days** |

yishunlu-222, 2023-12-11:
Hi Danny,

I'm DPhil student from Oxford and I have a project with Diamond Light Source.
It is very exciting to see someone who can reproduce the analytical absorption correction. This is because I failed to achieve that. Would you mind showing me the code?

Kind regards,
Yishun

DannyHindson, 2023-12-15:
> Hi Danny,
> 
> I'm DPhil student from Oxford and I have a project with Diamond Light Source. It is very exciting to see someone who can reproduce the analytical absorption correction. This is because I failed to achieve that. Would you mind showing me the code?
> 
> Kind regards, Yishun

Hi Yishun,

I've changed roles since I created this GitHub issue (still working at STFC but no longer working in ISIS on the Mantid project). The instrument scientist team working on the SXD experiment at ISIS wrote some C or C++ code that implemented the method described in the Tompa paper that is linked on this issue. The aim of this GitHub issue was to port that code into Mantid. As far as I'm aware that's not been done yet. So the best that we could do is see if we could share the SXD implementation with you. Since it's not actually my code I'm a bit hesitant about sharing with you directly. I think it might be best if you approached someone working on the SXD experiment directly (perhaps Silvia Capelli silvia.capelli@stfc.ac.uk) and asked her if she'd mind sharing the code. Are you happy to contact her? Feel free to mention my name or Richard Waite who is still working in the Mantid team

Thanks
Danny