mantidproject/mantid #34787: Populate OutputWorkspace with LHS workspace instead of RHS if possible

robertapplin, 2022-11-29:
**Original reporter:**
Liliana De Campo

**Describe the bug**
Population the output workspace name in an algorithm dialog with a LHS and RHS workspace will populate using the RHS workspace. It probably makes more sense to populate using the LHS workspace.

**To Reproduce**

For example
1. Open the Minus algorithm dialog
2. Enter a LHS and RHS workspace
3. Click the populate output workspace name button
4. It populates using the RHS workspace, but it probably makes more sense to use the LHS workspace.

**Expected behavior**
The LHS workspace name should be copied instead of the RHS workspace.

**Screenshots**
[image]

robertapplin, 2022-11-29:
@thomashampson would this be a core issue?

jhaigh0, 2023-06-21:
Doesn't seem to be a consistent rule...

[image]

jhaigh0, 2023-06-23:
Okay, I've just had this happen with `Plus` so still needs to be understood better.