mantidproject/mantid #34794: Filtering logs re-design

peterfpeterson, 2022-11-29:
**Is your feature request related to a problem? Please describe.**

The correct solution will give the correct results for (being python-centric) the `mantid.api.Run` to give back correct results for `getProtonCharge()` (which should be a sum of the `proton_charge` log) and `getPropertyAsSingleValueWithTimeAveragedMean()` and `getTimeAveragedStd()`. 

**Describe the solution you'd like**

The solution is to introduce a new object, `TimeMask` which contains information about when the workspace was active. This is in contrast to the current implementation which copies parts of the logs and adds artificial log values at the boundaries of when the workspace is active and not active.

The new object, `TimeMask` will be a specialization of either `TimeSeriesPropertyBool` or `TimeSeriesPropertyInt` depending on whether it can be used for filtering the events themselves as well. Either way, `0` or `false` will mean that the time period is to not be included and `>0` or `true` means that the time period should be included.

The `Run` object will get a `TimeMask` added to it so it can return the 

**Describe alternatives you've considered**

**Additional context**

This is associated with:
* mantid roadmap item [SANS time-slicing performance improvements #24 ](https://github.com/mantidproject/roadmap/issues/24)
* [ornl gitlab epic](https://code.ornl.gov/sns-hfir-scse/sans/sans-backend/-/issues/882)

peterfpeterson, 2022-12-06:
I appreciate that you're looking into how the `TimeMask` would actually work. A lot of these cases should be turned into unit tests for `Run::timeAverageMean(const string &)`

AndreiSavici, 2022-12-07:
In addition, some unit, docs, or system tests might fail. We should keep track of these, and see if the changes make sense (don't just use new numbers for the output).

peterfpeterson, 2022-12-07:
Review Notes:
* Need to sort out what `Kernel::LogFilter` is. It appears to only be used in `DataHandling::ISISRunLogs` and may be a similar concept to a `TimeSplitter` or `TimeMask`.
*  Need to sort out how to merge workspaces, specifically, adding two previously filtered workspaces

martyngigg, 2022-12-21:
I generally like the concept here. The current filtering is quite confusing to understand. A question about the high-level changes:

> High level changes

    filtered logs will be copies of the original: no extra or removed values

It feels like the filtered log should just be a view onto the original data. Do we need to copy or can we just reference to save even more memory?

peterfpeterson, 2022-12-21:
That is a good idea. It would be facilitated by the handles on the logs being copy-on-write pointers.

mantid-builder, 2023-01-19:
Fixed by #34938

PR #34938 (2023-01-18): Tests for time averages
**Description of work.**

As part of #34794:

- [x] overload `TimeSeriesProperty::timeAverageValue`
- [x] expose `TimeSeriesProperty::timeAverageValue(const TimeROI &timeRoi) const` to Python

**To test:**
```python
# Here code to carry out a test with property.timeAverageValue(time_roi)
```

This does not require release notes because release notes will be written with the PR that will finally close #34794

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review ### [... cut]
closed 2023-01-19 by mantid-builder