mantidproject/mantid #34798: C++ version of Meier

AnthonyLim23, 2022-11-30:
The Meier function is currently written in Python. It has had some problems due to the spin value only being able to take half integer values causing fits to behave erratically. Looking at the original code it would be better suited to C++ for performance. This has 2 main parts: 

- Rewrite the code to C++
- Change spin from being a parameter to an attribute

AnthonyLim23, 2022-11-30:
[image] [C++ version of Meier](https://trello.com/c/iROfzvyd/31-c-version-of-meier)

MohamedAlmaki, 2022-12-20:
I just catch this bug when I was reimplementing the function, see [line 68](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/plugins/functions/Meier.py#L68).

MohamedAlmaki, 2022-12-23:
I've implemented and refactored the function in C++. Also, changed the spin from a parameter to an attribute. But in I'm not sure how to implement the test_do_fit unit test in C++ (see [test](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/test/python/plugins/functions/MeierTest.py#L31)). There is no equivalent help function in C++ like in Python to do this unit test. Also, there is no C++ function that is tested using a fit unit test. Most unit tests in C++ are used to just test the input and output of the function.

AnthonyLim23, 2023-01-04:
> I just catch this bug when I was reimplementing the function, see [line 68](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/plugins/functions/Meier.py#L68).

what bug was this?

AnthonyLim23, 2023-01-04:
> I've implemented and refactored the function in C++. Also, changed the spin from a parameter to an attribute. But in I'm not sure how to implement the test_do_fit unit test in C++ (see [...]

For "simple" functions we can easily calculate what the result should be. So can test the function evaluation directly. For more difficult examples it can be difficult. However, in this case you probably want to check that the spin has not changed value.

Have you tried calling the fit algorithm in the test?

MohamedAlmaki, 2023-01-04:
> > I just catch this bug when I was reimplementing the function, see [line 68](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/plugins/functions/Meier.py#L68).
> 
> what bug was this?

Sorry, I mean [line 64](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/plugins/functions/Meier.py#L64). It should be "append" not "apend"

PR #34912 (2023-04-17): Reimplement Meier function using C++
**Description of work.**
This PR adds a refactored C++ implementation of the Meier function and its unit tests.

**To test:**
Open the fit browser and search for the "MeierV2" function in the functions list. Then, you can edit the parameters of the function and run the fit. Make sure that the function behaves as expected.

Fixes #34798. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the cod [... cut]
closed 2023-04-17 by peterfpeterson