mantidproject/mantid #34820: IDAaaS Freeze and crash when Browsing to Directory with lots of files

robertapplin, 2022-12-06:
**Original reporter:** Sarah Rogers, Adhip Rahman

**Describe the bug**
Browsing to a directory from the Manage User Directories dialog on IDAaaS can sometimes freeze Mantid. Then clicking on the X of the dialog causes Mantid to crash. This seems to happen when browsing to directories that contain a large number of files.

**To Reproduce**

For example
1. Create a Mantid workspace on IDAaaS here https://dev.analysis.stfc.ac.uk/
1. Open ISIS SANS interface
2. Click `Manage User Directories`
3. Click `Browse To Directory`
4. Browse to the following directory `/archive/NDXLARMOR/Instrument/data/cycle_22_3/`. Or another directory containing a large number of files
5. Mantid should freeze.
6. Click on the `X` for the Browse dialog and Mantid will crash

**Expected behavior**
There should be no freeze if possible, and no crash.

**Platform/Version (please complete the following information):**
 - IDAaaS

thomashampson, 2023-02-23:
Can we confirm that this is a linux only problem and also that it's not specific to IDAaaS?

MialLewis, 2023-02-23:
I'm sure this is a not very good test, but I copied the `/archive/NDXLARMOR/Instrument/data/cycle_22_3/` folder onto my hard drive and can access it without hang using `WSL`. If anyone can check with an actual linux machine that would be great.

thomashampson, 2023-03-22:
Thanks to @samtygier-stfc for helping track down a fix for this, see here:
https://github.com/mantidproject/mantidimaging/issues/1742
To summarise:
Mantid imaging's nightly suffers from the same issue since they moved the pyqt5 package installing from pypi to conda-forge. This is because the conda pacakges do no include the GTK integration. See these:
https://github.com/conda-forge/pyqt-feedstock/issues/87
https://github.com/conda-forge/qt-main-feedstock/issues/120
We should be able to fix it by installing the `qt-gtk-platformtheme` package:
https://github.com/conda-forge/qt-gtk-platformtheme-feedstock

This requires us upgrading qt-core to 5.15.8, which we had pinned back to 5.15.4 due to QTabWidget rendering issues in the instrument viewer and some scientific interfaces (https://github.com/mantidproject/mantid/pull/35013)

closed 2023-05-04 by peterfpeterson