mantidproject/mantid #34883: SANS GUI: Load warning/error should not be suppressed

DanielMurphy22, 2022-12-20:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

gemmaguest, 2022-12-20:
I've just tried this in the old SANS interface in MantidPlot v2.6.0 and it loads the raw data instead of giving an error. The new interface is more picky/specific about which loader it uses though - if it looks like the user wants to load nexus files then it will directly call the correct loader e.g. LoadISISNexus, so will not attempt to load raw data unless the user specifically asks it to. It would be quite a bit of rework to change this so I think the suggestion to provide a better help message is the best option (and if we can make this more obvious somehow we should do that too).

smk78, 2022-12-21:
I'm content with us needing to enter <runnumber>.raw in the GUI table if we specifically want to load Raw data files, and for the loader to assume it needs to look for .nxs if we don't specify an extension. That's ok.

But the other aspect here is why can't it load the 2007 NeXus data files?

gemmaguest, 2022-12-22:
> I'm content with us needing to enter .raw in the GUI table if we specifically want to load Raw data files, and for the loader to assume it needs to look for .nxs if we don't specify an extension. That's ok.
> 
> But the other aspect here is why can't it load the 2007 NeXus data files?

Ok, it's a bit hard to be certain without having a working version of Mantid from that time (which would probably involve a lengthy and difficult build!), but as far as I can tell we have never been able to load them; here's what I've found:
- The format changed in cycle 09_1 between runs 49885 (2009-04-27, fails) and 49886 (2009-04-28, works).
- The main difference I spot is that the top level tag is named `raw_data` in the former and `raw_data_1` in the latter.
- The LoadISISNexus algorithm was added around the same time (2009-05-11) using the `raw_data_1` tag; 49886 was subsequently added as unit test data for it.
- Prior to this, there was a LoadNeXus algorithm, but from looking at the code it seems to only load muon data and processed nexus. I can't find any references to the code reading a `raw_data` tag anywhere, either from then or now.

It does look like the format was being settled upon at that time:
- Cycle 08_1 has some .nxs files but subsequent cycles in 08 don't have .nxs files
- CRISP follows a similar pattern with .nxs files in 08_1 but not the rest of 08. It has a top-level tag of `run` then moves to `raw_data_1` at the start of 09_1.

smk78, 2022-12-22:
Ok. That tracks.

TS2 commenced ops in 2008 and SANS2D run 00000001 was 2009-03-05 but a RAW file. The first NXS was 00000020 on 2009-04-17. It was SANS2D that _needed_ Mantid, whilst we ran Mantid and COLETTE in parallel on LOQ for quite some time (a good couple of years). But I couldn't remember _when_ LOQ switched over. LARMOR and ZOOM didn't start producing data until 2013 and 2016, respectively. I do recall Muons 'pioneering' NeXus.

So... I think all we need is some conditional tests and an appropriate message ('cant load this old NeXus, use Raw instead'), isn't it? If fixing in the SANS GUI, then only for LOQ & SANS2D. If fixing in LoadISISNexus itself then it'll have to be for a bunch of instruments. @FreddieAkeroyd should know the details of dates/format change.

closed 2023-10-16 by github-actions[bot]