mantidproject/mantid #34943: Manual Testing ISIS Indirect interfaces

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow the directions at: https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

Harrietbrown, 2023-01-16:
ISIS calibration bin widths for the resolution need to be exact, this means the dragging range can not work as the result can't be exact.

Harrietbrown, 2023-01-16:
colourfill plots are spilling over the boundaries and coveringt he z axis image.

Harrietbrown, 2023-01-16:
Conv fit plot output for all now produces 5 plots as it includes chi_sq

Harrietbrown, 2023-01-16:
conv fit plot guess has a different function to the fit functuion after run is clicked.

martyngigg, 2023-01-17:
@Harrietbrown Could you create an issue/separate issues for those things found (except for "colourfill plots are spilling over" as that is captured by the core testing) so that we can close this as complete?

Harrietbrown, 2023-01-23:
Issue made for the docs update (#35063) and the ISIS Calibration bin widths issue (#35041) I was unable to replicate the issue with the guess in conv fit.

closed 2023-01-23 by Harrietbrown