mantidproject/mantid #34944: Manual Testing Documentation

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Check Online Docs and the Qt-help docs built into MantidWorkbench (from the help droppdown menu)

* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

You may wish to use this script ( [OpenMostDocumentationForTesting.py](https://github.com/mantidproject/mantid/blob/master/tools/scripts/OpenMostDocumentationForTesting.py) ) to open all the online docs pages!

sf1919, 2023-01-16:
Testing notes need to be expanded - not very clear to allow testing

sf1919, 2023-01-16:
On Algorithm list of all algorithms there are two separate P sections on Mantid Workbench which is confusing. There seems to be two columns on the page on the internet.

sf1919, 2023-01-16:
Random crash when scrolling through an algorithm page on help in Workbench. We raised an error report.

sf1919, 2023-01-16:
Search box is not working in Workbench (ILL conda install on Ubuntu 18 - might be the proxy issue)

sf1919, 2023-01-16:
Latex not showing on website help (but comparison page does render properly within Workbbench) - https://docs.mantidproject.org/nightly/algorithms/PolarizationCorrectionWildes-v1.html

sf1919, 2023-01-16:
Trying to print to pdf from help within Mantid Workbench crashes Mantid. Submitted error reporter report.

MialLewis, 2023-01-16:
> Testing notes need to be expanded - not very clear to allow testing

Issue #34970

MialLewis, 2023-01-17:
> Search box is not working in Workbench (ILL conda install on Ubuntu 18 - might be the proxy issue)
> 
> Not reproducable on Windows or IDAaaS

Requested add to #33969

MialLewis, 2023-01-17:
> Search box is not working in Workbench (ILL conda install on Ubuntu 18 - might be the proxy issue)
> 
> Not reproducable on Windows or IDAaaS

Will try and replicate this on Ubuntu outside of ILL before raising, though nightly currently not working with WSL...

MialLewis, 2023-01-17:
> On Algorithm list of all algorithms there are two separate P sections on Mantid Workbench which is confusing. There seems to be two columns on the page on the internet.
> 
> Same with F.

#34972

MialLewis, 2023-01-17:
Raised all related issues, one outstanding waiting on further reproduction.

closed 2023-01-17 by MialLewis