mantidproject/mantid #34945: Manual Testing Workbench Group 1- 4

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

MohamedAlmaki, 2023-01-16:
When loading a file with a lot of parameters, the window will be stretched and not centered. the file name used in the test is CNCS_7860_event.nxs
[image]

DanielMurphy22, 2023-01-16:
On instructions, mention that About MantidWorkbench menu is not under help on MacOS but under the application name.

DanielMurphy22, 2023-01-16:
Should move the privacy policy from the archive page to the main docs.

thomashampson, 2023-01-16:
The hover-over cursor changes in the fit browser are not behaving completely as expected. If you hover over a line and then move upwards off the plot it will stay as the horiztonal resize cursor.
[image]

DanielMurphy22, 2023-01-16:
Colorfill and surface (and contour plots) have the colorbar in the wrong place: `This figure includes Axes that are not compatible with tight_layout, so results might be incorrect.`
[image]

DanielMurphy22, 2023-01-16:
[image]
If I draw a zoom box on sliceviewer or on a 1D plot (here in the bottom right), then the displayed zoom box is in the wrong location on MacOS, but the region it zooms to is the intended region of the user.

DanielMurphy22, 2023-01-16:
Not a bug, but worth noting: Project save does not include sliceviewer or the fit property browser on plots

DanielMurphy22, 2023-01-16:
`The to_draw attribute was deprecated in Matplotlib 3.5 and will be removed two minor releases later.` when drawing a region in sliceviewer

DanielMurphy22, 2023-01-16:
Sliceviewer with region selection region, if you change the normalisation then the lineplots stop displaying.

thomashampson, 2023-01-16:
I am struggling to reproduce this, but after closing the sliceviewer and opening a 1D plot, I get this message:

```
Traceback (most recent call last):
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/matplotlib/cbook/__init__.py", line 307, in process
    func(*args, **kwargs)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/views/cursor.py", line 57, in on_mouse_leave
    self.on_cursor_outside_axes()
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/presenters/imageinfowidget.py", line 67, in on_cursor_outside_axes
    self._widget.cursorAt(DBLMAX, DBLMAX, DBLMAX, extra_cols)
RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
```

MohamedAlmaki, 2023-01-16:
The gradient bar should be out of the plot:
[image]

gui-co, 2023-01-16:
On plots, the distance between marker and label changes with the axis scale (lin <-> log):

[image]

[image]

thomashampson, 2023-01-16:
Attempting to show or hide all errors on a waterfall plot will casue unhandled exception
[image]

DanielMurphy22, 2023-01-16:
Plotting with a script from here: https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/plotting/01_basic_plot_scripting.html#d-surface-and-wireframe-plots 
script:
```python 
from mantid.simpleapi import *
import matplotlib.pyplot as plt

data = Load('MUSR00015189.nxs')
data = mtd['data_1'] # Extract individual workspace from group

fig, ax = plt.subplots(subplot_kw={'projection':'mantid3d'})
ax.plot_surface(data)
plt.show()
```

I get an empty plot window and the following error message:
```
Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/backends/backend_qt.py", line 454, in _draw_idle
    self.draw()
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/backends/backend_agg.py", line 405, in draw
    self.figure.draw(self.renderer)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/artist.py", line 74, in draw_wrapper
    result = draw(artist, renderer, *args, **kwargs)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/figure.py", line 3071, in draw
    mimage._draw_list_compositing_images(
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mpl_toolkits/mplot3d/axes3d.py", line 440, in draw
    for artist in sorted(collections_and_patches,
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mpl_toolkits/mplot3d/axes3d.py", line 441, in <lambda>
    key=lambda artist: artist.do_3d_projection(),
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mpl_toolkits/mplot3d/art3d.py", line 800, in do_3d_projection
    cface = cface.repeat(len(xyzlist), axis=0)
AttributeError: 'tuple' object has no attribute 'repeat'
```

MohamedAlmaki, 2023-01-16:
When executing a script that plots some figure, all hidden plots will be shown with the figure executed from the script. Also, if the figure is already present in the hidden menu, it will be plotted with extra legend
[image]

DanielMurphy22, 2023-01-16:
QApplication event loop is already running message is printed to terminal when script has run as we run scripts in the script editor in a separate thread.

DanielMurphy22, 2023-01-16:
type 0,003 in as the max limit on the colorbar in the sliceviewer.
```
        Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/colorbar/colorbar.py", line 261, in clim_changed
    self.update_clim()
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/colorbar/colorbar.py", line 270, in update_clim
    self._manual_clim()
  File "/Applications/MantidWorkbenchNightly.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/colorbar/colorbar.py", line 338, in _manual_clim
    cmax = float(self.cmax.text())
ValueError: could not convert string to float: '0,003'
```

thomashampson, 2023-01-16:
Attempting to fix a parameter in the fit browser will cause a hard crash without any stack trace. MacOS reveals this useful information:
```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000048
Exception Codes:       0x0000000000000001, 0x0000000000000048
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [38588]

VM Region Info: 0x48 is not in any region.  Bytes before following region: 140737488084920
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7ffffffbe000-7ffffffbf000 [    4K] r-x/r-x SM=ALI  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libMantidAPI.dylib            	       0x1078bcbf4 Mantid::API::ParameterTie::asString(Mantid::API::IFunction const*) const + 52
1   libMantidQtWidgetsCommonQt5.dylib	       0x15de8687c MantidQt::MantidWidgets::FitPropertyBrowser::stringChanged(QtProperty*) + 1180
2   libQt5Core.5.15.6.dylib       	       0x10fd33ce4 0x10fbe8000 + 1359076
```

thomashampson, 2023-01-16:
Somehow (not sure yet), it is possible to cause an unhandled exception when a plot is open with the fit browser open. Resizing it, maybe squeezing it to get a tight layout warning?
```

Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\backends\backend_qt.py", line 272, in enterEvent
    *self.mouseEventCoords(event),
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\backends\backend_qt.py", line 265, in mouseEventCoords
    x = pos.x()
AttributeError: 'QPaintEvent' object has no attribute 'x'

Maybe related to this:
https://github.com/matplotlib/matplotlib/issues/11607
```

gui-co, 2023-01-16:
> Somehow (not sure yet), it is possible to cause an unhandled exception when a plot is open with the fit browser open. Resizing it, maybe squeezing it to get a tight layout warning?
>
> ```
>
> Traceback [...]

This looks similar:
https://github.com/matplotlib/matplotlib/issues/11607

DanielMurphy22, 2023-01-16:
drag and drop file loading does not work on MacOS
[image]

MohamedAlmaki, 2023-01-16:
When I try to load a CSV file, I get this error
```
Failed to read as ASCII this file: 'C:/Users/joy22959/Documents/samples/UsageData/input_csv_example.csv, error description: No valid data in file, check separator settings or number of columns per bin.
Error in execution of algorithm LoadAscii:
Failed to recognize this file as an ASCII file, cannot continue.
Error in execution of algorithm Load:
Failed to recognize this file as an ASCII file, cannot continue.
```

DanielMurphy22, 2023-01-17:
https://github.com/mantidproject/mantid/issues/34081

thomashampson, 2023-01-17:
Algorithm history is a bit broken for Mean algorithm, try using it on two workspaces and the "Plus" step has the incorrect LHC workspace:
```
from mantid.simpleapi import *

Load(Filename='C:/workdir/UsageData/MAR11015.raw', OutputWorkspace='MAR11015_b')
Load(Filename='C:/workdir/UsageData/MAR11015.raw', OutputWorkspace='MAR11015_a')
Plus(LHSWorkspace='output', RHSWorkspace='MAR11015_b', OutputWorkspace='output')
CreateSingleValuedWorkspace(OutputWorkspace='__python_binary_op_single_value', DataValue=2)
Divide(LHSWorkspace='output', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='output')
Mean(Workspaces='MAR11015_a, MAR11015_b', OutputWorkspace='output')
```

MohamedAlmaki, 2023-01-17:
When you open the same script in mantid, it will be open in a new tab and not in the same tab.

MohamedAlmaki, 2023-01-17:
Also, changes outside mantid in some script will not be updated automatically in mantid editor.

gui-co, 2023-01-17:
With an AZERTY keyboard, the keyboard shortcut to comment/uncomment in the script editor (Ctrl+/) is not working.

gui-co, 2023-01-17:
In the script editor with the search bar opened (Ctrl+F), the focus is kept by this search bar. If one tries to press enter to execute an algorithm for example, it does not work.

thomashampson, 2023-01-17:
Changing a tie in the fit GUI to a number will tell you that it's not possible but not revert to the correct value.

DanielMurphy22, 2023-01-20:
All issues raised separately

closed 2023-01-20 by DanielMurphy22