mantidproject/mantid #34946: Manual Testing Muon

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

martyngigg, 2023-01-17:
In https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html#ionic-diffusion-test the output we obtained `4.379e-8` for `f1.Nu` and not the listed number.

martyngigg, 2023-01-17:
We were unable to complete the second EMU test, `Superconducting Copper Test`. The interface seems to freeze whenever you try to load a number of runs at the same time.

martyngigg, 2023-01-17:
We paused testing as all loading into `MuonAnalysis` seems to freeze workbench when loading a range of runs so we were unable to get any further.

AnthonyLim23, 2023-01-20:
I couldn't install the windows nightly, so I am using the nightly on IDAaaS.

For the ionic diffusion test I got a nice plot:
[image]
@martyngigg I think a setting may have been missed, its hard to say without seeing the interface.

The superconducting copper test seems to be ok too. Maybe the data archive was down?
[image]

martyngigg, 2023-01-23:
@AnthonyLim23 If you're able to produce the desired output then that must have been tester error.

The other issues must have been the archive playing up as you say.

jhaigh0, 2023-01-23:
[developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html](https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html)
 - An instruction to explain feature flags is needed since they are necessary for one of the tests
 - We needed to change the fit peak centre for the last step of the super-conducting copper test for the fit to be sensible
 - On the HIFI Transverse Field Simultaneous Fitting test, our first fit had Chi > 100 but still displayed as a success. Also it should say Chi Squared rather than just Chi in the testing doc
 - @rbauststfc Experienced the interface hiding itself when changing instrument (although this wasn't a problem for @MialLewis)
 - In the Transverse Field Asymmetry test, TF Asymmetry Mode is in a combo box, not a check box in the bottom table as described

[ALC](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_ALC.html)
 - After following the first test instructions, we ended up with the peak below; not sure if it is what is expected
 
[image]

[PSI](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_PSI.html)
 - We could not find the file dolly 1529 - better instructions detailing where / how to access it are needed

[HIFI](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_HIFI.html)
 - In the second test, no fitting function is added before running a sequential fit

jhaigh0, 2023-01-24:
> > [developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html](https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html)
> > 
> > * An instruction to explain feature flags is needed since they are necessary for one of the tests
> > * We needed to change the fit peak centre for the last step of the super-conducting copper test for the fit to be sensible
> > * On the HIFI Transverse Field Simultaneous Fitting test, our first fit had Chi > 100 but still displayed as a success. Also it should say Chi Squared rather than just Chi in the testing doc
> > * @rbauststfc Experienced the interface hiding itself when changing instrument (although this wasn't a problem for @MialLewis)
> > * In the Transverse Field Asymmetry test, TF Asymmetry Mode is in a combo box, not a check box in the bottom table as described
> > 
> > [ALC](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_ALC.html)
> > 
> > * After following the first test instructions, we ended up with the peak below; not sure if it is what is expected
> > 
> > [image]
> > [PSI](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_PSI.html)
> > 
> > * We could not find the file dolly 1529 - better instructions detailing where / how to access it are needed
> > 
> > [HIFI](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_HIFI.html)
> > 
> > * In the second test, no fitting function is added before running a sequential fit
> 
> For ALC: I got a nice fit following the instructions. [image] Looking at the plot, I think the baseline modelling has gone a bit wrong.
> 
> For PSI: who should I send the data to? It should also be in the test data
> 
> For HIFI: the fit function does not matter, its just to demonstrate the fitting and multiple periods. A linear BG and a ExpDecayMuon work ok.

Ah, that's what we thought the plot should've looked like, must have made a mistake.

We found an md5 file in the testing data, but the interface was expecting a nxs file (I think). You can send me the file and I can pass it onto the windows testers.

The issue for HIFI was that there was no instruction to add a fitting function, which was initially confusing, and we thought it would be easier to follow if there was an additional line reading 'Add a fitting function of your choice' (or something similar).

ConorMFinn, 2023-01-24:
PSI:

Occasional crash when changing to flat background/clicking Use Raw in corrections. Seems to be when algorithms complete and RemoveFInishedAlgs is called.

Couldn't get beyond these crashes so testing is incomplete. 

@AnthonyLim23 This is a regression from 6.5. Is this something that needs fixing for the release?

AnthonyLim23, 2023-01-24:
I will have a look this afternoon to see if I can reproduce it. If I have trouble reproducing it I will call @ConorMFinn 

I cannot think of any reason why this bug would be added (that code has not been touched by the spec team for about a year).

ConorMFinn, 2023-01-24:
I expect it's probably a higher issue to do with how we destroy algorithms that just happens to get picked up here.

MialLewis, 2023-01-24:
It doesn't look like that's changed recently either. It looks to be an access violation when clearing `PropertyManager::m_properties` upon the destruction of `PropertyManager`, the destruction of which is cascading from `AlgorithmManagerImpl::removeFinishedAlgorithms`.

`m_properties` is a map of unique pointers, so I imagine a property one of these pointers points to has been deleted elsewhere?

MialLewis, 2023-01-27:
Issue raised here: #35096

AnthonyLim23, 2023-01-31:
the doc changes have been copied over and the bug has been fixed, so I am closing this.

closed 2023-01-31 by AnthonyLim23