mantidproject/mantid #34950: Manual Testing Workspace Groups

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Follow the examples at http://docs.mantidproject.org/concepts/WorkspaceGroup.html 
* Ensure MantidWorkbench displays Group Workspaces correctly in the workspace tree

RichardWaiteSTFC, 2023-01-17:
No new issues - could reproduce the issues found last manual testing
1. Can add a workspace group to a workspace group using a script #34366 
2. Can add the same workspace to two or more groups
```
ws1 = CreateSampleWorkspace()
ws2 = CreateSampleWorkspace()
ws3 = CreateSampleWorkspace()

wsGroup1 = GroupWorkspaces("ws1,ws2")
wsGroup2 = GroupWorkspaces("ws1,ws3")
```
Point (2) might not be an issue, it seems not to copy the workspace and deleting the workspace deletes it from both etc.

RichardWaiteSTFC, 2023-01-17:
Also maybe this could be turned into a doc-test?

closed 2023-01-17 by RichardWaiteSTFC