mantidproject/mantid #34952: Manual Testing ISIS Reflectometry

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

sf1919, 2023-01-16:
On Ubuntu 18 using nightly direct from Conda.

When doing Polarisation Corrections test we sometimes find when switching between Runs tab and the Instrument Setting tab that the Instrument Settings keep appearing within the Runs tab. We think the GUI is not refreshing correctly.

perenon, 2023-01-16:
[Missing Test] the preview tab testing is not included is the manual testing instructions.

rbauststfc, 2023-01-16:
I've raised separate issues for the problems we found, so this can now be closed.

closed 2023-01-16 by rbauststfc