mantidproject/mantid #34953: Manual Testing ISIS SANS

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

robertapplin, 2023-01-18:
1. Change the Radius maximum setting
2. Edit the table in Run tab
3. Export as batch file
4. Open the setting tab
5. The radius maximum has been reset

robertapplin, 2023-01-18:
1. Change the Radius Maximum in the Beam centre tab to 360.
2. Reload the user file
3. Radius maximum in the beam centre tab doesn't get reset. Is this expected?

robertapplin, 2023-01-18:
We think instructions 4 & 5 in the Display Mask section is the wrong way around. And 7 & 8 in the Processing -1D reduction instructions are the wrong way around (in the manual testing instructions)

MohamedAlmaki, 2023-01-18:
In Sum Runs test, instruction 3 the output file is already prefilled so you don't need to add it

MohamedAlmaki, 2023-01-18:
In the settings tab->q, wavelength, detector limits table->wavelength section there are missing tooltips. 
[image]

ConorMFinn, 2023-01-25:
> It is not a regression.
> 
>     1. Change the Radius maximum setting
> 
>     2. Edit the table in Run tab
> 
>     3. Export as batch file
> 
>     4. Open the setting tab
> 
>     5. The radius maximum has been reset

This one is a known issue (that might have one already, not sure). Making changes to the runs page reloads all of the settings from the user file. It's very annoying, but the scientists know about it so as far as I know they work around it.

ConorMFinn, 2023-01-25:
> It is not a regression.
> 
>     1. Change the Radius Maximum in the Beam centre tab to 360.
> 
>     2. Reload the user file
> 
>     3. Radius maximum in the beam centre tab doesn't get reset. Is this expected?

Fairly sure these come from the instrument parameter file so they don't get overwritten on a user file reload.

ConorMFinn, 2023-01-25:
> Is this a bug? It is not a regression. In the settings tab->q, wavelength, detector limits table->wavelength section there are missing tooltips. 

Yeah, that's a mistake. Good catch.

DanielMurphy22, 2023-01-25:
> https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

I think this came about when we ran the instructions under the section User Files: https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html#user-files If some of the Beam centre details shouldn't be reset when reloading the User file, are we able to clarify this in the testing instructions? Should I add a comment about this to https://github.com/mantidproject/mantid/issues/35051 ?