mantidproject/mantid #34954: Manual Testing Live Data ISIS

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html

robertapplin, 2023-01-17:
When the path to the `fileeventdatalistener.filename` property is wrong, and you try to select the `ADARA_FileReader` instrument in Load LiveData, Mantid crashes

robertapplin, 2023-01-17:
The instructions in the 'ADARA Fake Event' could do with an explicit description on how to start collecting live data

DanielMurphy22, 2023-01-20:
Issues raised