mantidproject/mantid #34955: Manual Testing Mantid Basics Course

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Check the Qt-help docs, built into Workbench, against the online documentation 
* Open up the basic course: http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course
* Check that the information and exercises are up to date and work in the Workbench release candidate

sf1919, 2023-01-17:
Displaying 2D Data part of the course needs updating - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data

It does not have the correct image or any information relating to the cutting tool added in PR #33765

sf1919, 2023-01-17:
Plot tool bar graphic needs updating - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html#formatting-plots

It does not include the new hide button introduced in PR #34614

sf1919, 2023-01-17:
The scripting example on this page is out of date - https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#basic-1d-plots

Legend has no attribute dragable

sf1919, 2023-01-17:
Took many attempts to download the Training course data from Sourceforge

rbauststfc, 2023-01-17:
The Toolbar buttons on this page need updating - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#the-toolbar

rbauststfc, 2023-01-17:
Need to check steps for zooming into mini-plot in previous Mantid version, as in beta the mini plot disappears as soon as mouse moves off detector - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#picking-a-single-detector

rbauststfc, 2023-01-17:
Screenshot showing mini plot context menu includes option `Save plot to workspace` that doesn't seem to be there any more - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#integrating-over-the-time-of-flight

rbauststfc, 2023-01-17:
We can't seem to replicate the behaviour described in this part of the course - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/04_masking_and_grouping.html#masking

DanielMurphy22, 2023-01-19:
Sample logs page, the first screen shot is of BSS_11841_event rather than CNCS_7860_event https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_sample_logs.html#sample-logs

DanielMurphy22, 2023-01-19:
Add `ENGINX00228061_log_alarm_data.nxs` to the TrainingCourseData as referenced here: https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/07_invalid_sample_logs.html

DanielMurphy22, 2023-01-19:
Instrument view drop down menus appear far away from the original combobox. This a can be quite variable and changes as you move and resize the interface[image]

DanielMurphy22, 2023-01-19:
02_complex_case the following paragraph is confusing as it mentions dragging on the plot after having set a FlatBackground, but this info should really be for the peak lines.

> Note that two fit functions have appeared in the Fit Property Browser called f0-Gaussian and f1-FlatBackground. Click on the triangles beside these functions to reveal their fit parameters. Just as with the StartX/EndX values, dragging on the plot and changing these values update each other.

DanielMurphy22, 2023-01-19:
Not sure the difference between Tie to function or Tie > Custom Tie. Tie to function doesn't seem to work in the cases we tried. Custom tie worked fine.

DanielMurphy22, 2023-01-19:
Qpaint error found when fitting: https://github.com/mantidproject/mantid/issues/35029

closed 2023-01-19 by rbauststfc