mantidproject/mantid #34957: Manual Testing Project Recovery

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

sf1919, 2023-01-16:
As we go through the tests when we restart Workbench we're getting at least two recoveries - the  last 2 in most cases. A bit confusing!

sf1919, 2023-01-16:
Test 4 - The Show Data Dialog does not come back after Project Recovery. The Reflectometry GUI also does not come back. The Engineering Diffraction GUI does come back though so it seems to be something specific to Show Data Table and the Reflectometry GUI.

sf1919, 2023-01-16:
First attempt at test 6 and Project Recovery options did not appear. Worked as expected 2nd time we attempted it.

sf1919, 2023-01-16:
Test 8 - the documentation may need updating for this one. By only choosing `Only open in script editor` then there is no workspace loaded that we can check the history of.

sf1919, 2023-01-16:
All relevant issues have been written up and linked to this issue for reference. Testing can now be closed.

closed 2023-01-16 by sf1919

rbauststfc, 2023-01-16:
> Test 4 - The Show Data Dialog does not come back after Project Recovery. The Reflectometry GUI also does not come back. The Engineering Diffraction GUI does come back though so it seems to be something specific to Show Data Table and the Reflectometry GUI.

I've just repeated this test and noticed that actually it worked OK, it's just that on Windows the Reflectometry GUI and Show Data dialog seem to be minimised so I must not have noticed that they were still there. I'll note this on the linked issues that were opened for the problem and close them.