mantidproject/mantid #34959: Manual Testing Sliceviewer

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

RichardWaiteSTFC, 2023-01-17:
Doc changes to manual testing:
Basic Usage:
- Matrix Workspace (2): Cut tool disabled for matrix workspaces (as well as non-ortho view etc.)
- Matrix Workspace (5b): Colorlimits are set to 0.1,1 (not 0,0)
- Matrix Workspace (6a): typo 'unchecked' should be 'checked'
- Matrix Workspace (7): Swap 7 with 6 for readability/ease of following
MD Workspaces:
- (1) Add test for HKL columns of cursor info table (and repeat in non-ortho view)
- Peak Overlay (2/3): Add line to test concise view (added this release)
Cutviewer:
- (4) Tool should now be enabled in non-orthog view
- (18) Sign of u1 & u2 should be flipped (elements are now positive)
- Add tests for non-ortho view cutview usage

RichardWaiteSTFC, 2023-01-17:
Issues found that were not regressions:
- [ ] Exporting ROI and cuts does throws error in MDHisto workspace (`'NoneType' object is non-subscriptable`)
- [ ] Exporting cuts of EventWorkspace also throws error (not a regression - see #33394)
- [ ] Deleting the original MDEvent workspace of an MDHisto open in sliceviewer does not close the sliceviewer (not a regression - see #33434)

RichardWaiteSTFC, 2023-01-17:
Issues to fix this release:
- [ ] After toggling the cut viewer tool on/off, when the image is rebinned (e.g. by changing nbins) the cut represention is drawn again (even though the tool is not enabled). Then in non-ortho view is enabled the the 1D plot widget of the cut tool opoens up on the right.
- [ ] Peak overlay in non-ortho view: clicking a peak doesn't navigate to the correct Q - think it is related to https://github.com/mantidproject/mantid/pull/34717/commits/2a2a8a24d69ca0066d7f9c59d18149b2542af9be
- [ ] Cursor info widget HKL columns code throws an out-of-range error for MD 4D workspaces

Other minor issues:
- [ ] Matplotlib deprecation warning when exporting cuts of MD workspace in sliceviewer:
`The to_draw attribute was deprecated in Matplotlib 3.5 and will be removed two minor releases later`
- [ ]  Moving cursor to last bin on RH edge of MDHisto workspace threw index error

RichardWaiteSTFC, 2023-01-17:
Issue found during core testing #34997

RichardWaiteSTFC, 2023-01-17:
All issues raised

closed 2023-01-17 by RichardWaiteSTFC