mantidproject/mantid #34960: Manual Testing Fit Script Generator

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html

Harrietbrown, 2023-01-17:
There are no spesific manual tests for this interface, only a usage example

Harrietbrown, 2023-01-17:
Instructions for second use instance mean that when we change the values of frequency to 1.3 it does not change the value in the first domain.

robertapplin, 2023-01-17:
- Full archive park in the usage example only works for windows. Remove the path and assume access to the dataset in a saved directory
- Try use one of the sample datasets as the usage example data
- clean up the setup script a little bit more.  (make it more readable). Shorten the titles of the workspaces to make it more readable in the plots

robertapplin, 2023-01-19:
Closing as a PR has been opened to fix these #35004

closed 2023-01-19 by robertapplin