mantidproject/mantid #34961: Manual Testing ALF

sf1919, 2023-01-16:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/DirectInelastic/ALFViewTests.html

robertapplin, 2023-01-16:
From testing we found the following things that might need investigating:

- The SymmetricLog10 setting is auto-selected when opening ALFView rather than Linear. Possibly picking up a setting from the InstrumentView?
- The dotted lines for the peak picker could be shown in grey when the peak picker is not selected. This will make it clearer what it is
- The peak picker could be given a background value so that it aligns better with the peak
- Need to check the precision needed for the rotation angle (and peak centre) as it differs slightly on each OS (only in the 6th decimal place)

martyngigg, 2023-01-17:
@robertapplin Could you make an issue for the things that need addressing so that we can close this?

robertapplin, 2023-01-18:
Issues have been created

closed 2023-01-18 by robertapplin