mantidproject/mantid #34964: Reflectometry GUI doesn't render properly when switching tabs

rbauststfc, 2023-01-16:
**Describe the bug**

The Reflectometry GUI doesn't seem to be refreshing correctly when switching between tabs, ultimately making the GUI unusable. We think this started after the Qt version update.

Found during manual testing  - https://github.com/mantidproject/mantid/issues/34952#issuecomment-1384147236

**To Reproduce**

1. Open the Reflectometry GUI
1. Switch between the Runs and Experiment Settings tabs a few times

**Expected behavior**

The GUI should refresh to display the correct widgets for each tab.

**Screenshots**

This is the view that eventually results and the GUI must be closed and re-opened to continue working:

[image]

**Platform/Version (please complete the following information):**

 We've seen this on Ubuntu 18 and IDAaaS. We haven't seen it on Windows and haven't yet tested on Mac.

ConorMFinn, 2023-01-16:
We think this might be Qt related, there's a similar issue in ALF view. 

It's not present on macOS

gui-co, 2023-01-17:
Similar issue with tabs of the instrument viewer and the global settings (File -> Settings).

(on Ubuntu 18)

RichardWaiteSTFC, 2023-01-17:
MIfght be related to an error I saw Opening ISIS Reflectometry > Reduction Preview
`OpenGL error detected in setRenderingOptions: invalid operation`
on windows with OpenGL in instrument view set to true in settings

gui-co, 2023-01-17:
In the script editor with the search toolbar open (Ctrl+F), the focus is kept by the search bar. If one tries to press "Enter" to execute an algorithm for example it does not work.

gui-co, 2023-01-17:
With an AZERTY keyboard, the keyboard shortcut used to comment/uncomment in the script editor (Ctrl+/) is not working.

ConorMFinn, 2023-01-17:
Qt5 on condaforge has been upgraded from 5.15.4 to 5.15.6 as of 2023-01-12. This is what was causing the issues, I just happened to have a slightly more out-of-date conda env than the nightly did. Suggest pinning over the release so testing of the refl GUI can continue.

mantid-builder, 2023-01-19:
Fixed by #35013

PR #35013 (2023-01-19): Fix Rendering Error 
**Description of work.**
Pinned qt on conda to a slightly earlier patch to avoid the rendering issues seen in the instrument viewer, alf, and refl guis. 

**To test:**
1. Update your conda environment from the supplied mantid dev `.yaml`
2. Launch workbench
3. Open the refl GUI & instrument viewer and flick through the tabs. 
4. Check that the rendering is working as it should. 

Fixes #34964 

*This does not require release notes* because **it concerns our conda management.**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.ma [... cut]