mantidproject/mantid #34967: Add manual testing documentation for the ISIS Reflectometry GUI preview tab.

rbauststfc, 2023-01-16:
We used the testing instructions [here](https://github.com/mantidproject/mantid/issues/31069) for manually testing the preview tab for the 6.6 release, but we need to update these and add them to the online manual testing documentation.

During [testing for the 6.6 release](https://github.com/mantidproject/mantid/issues/34952) we felt that the `INTER45455_inst` test data made it difficult to tell if moving the ROIs on the spectrum viewer was updating the plot (the reduction was clearly being re-run, but we concluded that the data changes must be so small that you couldn't see the plot changing). If possible, it would be good to find some test data that made this clear.

rbauststfc, 2023-01-25:
@gemmaguest I've written up some instructions for this but, before I put them into PR, I was wondering how best to make the `INTER45455_inst` test data available? It looks like the rest of the Reflectometry GUI testing data is either in the archive or in the ISIS sample data download. Would it be worth adding `INTER45455_inst` to the sample data and, if so, are there instructions anywhere for doing this?

gemmaguest, 2023-01-26:
We're in a slightly awkward situation at the moment because this run is on the archive but by default it will load with the incorrect IDF - so really it is the IDF that is missing rather than the data. So we could just provide the IDF - but to test the GUI it is useful to have it already loaded on the run so we can just enter the name/run, because part of the test is that it will load it - hence we've been using that custom version of the data with the correct IDF loaded on it.

Because this is fake data I think I'd prefer to keep it out of the sample data. I think it is in the unit test data, so for now we could just get developers to load it from there. I think it is named `INTER45455` so we need to be explicit because it could easily be confused with the one on the archive.

This isn't ideal, but longer term this problem will go away because when they have the real 2D detector the default IDF will be correct so we can start using real data.

rbauststfc, 2023-01-26:
Thanks @gemmaguest. I was just trying to play around with loading `INTER45455` from the unit test data, but the only way I could seem to do it was to load the hash file into the ADS. Is there another way this can be done that I'm missing?

PR #35131 (2023-02-16): Add ISIS Reflectometry GUI Preview Tab manual testing instructions
**Description of work.**

Adds manual testing instructions for the new preview tab in the ISIS Reflectometry GUI. We're using test data from the unit tests at the moment - the reason for this is explained on the linked issue and it shouldn't be necessary long-term.

**To test:**

Check that the new instructions make sense.

Fixes #34967. 

*This does not require release notes* because **it is an update to documentation.**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidprojec [... cut]
closed 2023-02-16 by DannyHindson