mantidproject/mantid #34979: Opening a python script twice will open two tabs

MohamedAlmaki, 2023-01-17:
**Describe the bug**

When you open a script in mantid that is already open in the editor, it will open in a new tab and not in the same tab.

**To Reproduce**
1- Open a script from some directory
2- The script will be opened in a new tab
3- Open the script again from the same directory
4- You will see a second tab is opened

**Expected behavior**
The script should be opened in the same tab.
**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 10 Enterprise
 - Mantid Version: 6.6.20230115.2237

**Additional context**

MohamedAlmaki, 2023-07-19:
@thomashampson @jhaigh0 
Do you think this issue is important?

jhaigh0, 2023-07-19:
I think it could be a good induction issue to go with #33369

thomashampson, 2023-07-20:
This would be a nice improvement

PR #35856 (2023-07-28): Fix bug that would open a duplicate tab if a file was already open
**Description of work**
Previously if you had a file open, then opened it again, then another tab would appear with the same file. What happens now is that the tab that's already open gets selected. If you have modifications in the tab that was open initially they won't be lost.

**Purpose of work**

Fixes #34979 

**Summary of work**

- Added two tests that check if you open a file a second time then you don't get a new tab. The second test checks that if the file is modified initially then it's still modified afterwards.
- Ch [... cut]
closed 2023-07-28 by gemmaguest