mantidproject/mantid #34980: Fit parameter ties don't revert when constant values are entered

thomashampson, 2023-01-17:
**Describe the bug**
In the fit browser, if you add a parameter tie, then attempt to change it to a constant value (i.e. a number), it will warn you but not revert to the previous value

**To Reproduce**
Steps to reproduce the behavior:

For example
1. Open a 1D plot
2. Open the fit panel
3. Add a Gaussian peak
4. Expand f0-Gaussian, and add a tie to the height with the value f0.Sigma, that should be fine
5. Try to change the Tie to a number, it will complain, but not revert to the previous correct value.

**Expected behavior**
Ideally, it would allow you to change a tie to a constant value. Otherwise it shoulf revert the value in the text box.

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: all
 - Mantid Version 6.5 and probably earlier

DanielMurphy22, 2023-01-25:
Here's some related info after poking around this bit of code. The mechanism for getting the oldTie, which I am looking at as part of #35081, seems to only return the Tie Expression in the case that it is a Tie to a function (e.g. f0.Height=f1.Height). If instead the oldTie was a custom tie (e.g. f0.Height=2.0) then the previous value is not found so an invalid change could not be reverted to.

thomashampson, 2023-05-18:
This has been resolved as part of the other bug fixes relating to fit parameter ties.

closed 2023-05-18 by thomashampson