mantidproject/mantid #34982: Crash when fixing fit parameter

thomashampson, 2023-01-17:
**Describe the bug**
Mantid will crash when you attempt to fix a fit parameter in the fit browser.

**To Reproduce**
1. Open a 1D plot
2. Open the fit panel
3. Add a peak
4. Expand the function, right click one of the parameters and click "Fix"

**Expected behavior**
Parameter should be fixed at current value without causing any problems.

**Platform/Version (please complete the following information):**
 - OS: all
 - Mantid Version 6.5.20230115.2237

**Additional context**
No stack trace, but the following output was taken from a crash on macOS:
```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000048
Exception Codes:       0x0000000000000001, 0x0000000000000048
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [38588]

VM Region Info: 0x48 is not in any region.  Bytes before following region: 140737488084920
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7ffffffbe000-7ffffffbf000 [    4K] r-x/r-x SM=ALI  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libMantidAPI.dylib            	       0x1078bcbf4 Mantid::API::ParameterTie::asString(Mantid::API::IFunction const*) const + 52
1   libMantidQtWidgetsCommonQt5.dylib	       0x15de8687c MantidQt::MantidWidgets::FitPropertyBrowser::stringChanged(QtProperty*) + 1180
2   libQt5Core.5.15.6.dylib       	       0x10fd33ce4 0x10fbe8000 + 1359076

```

AnthonyLim23, 2023-01-20:
I have had a quick look at this and the issue is in the GUI. The bug is happening when the fit property browser is meant to update to show the extra row underneath stating the fix. 

It is also worth noting that it is only when doing a fix. A tie is ok. In fact you can select tie, get the pop up box and enter `f0.A0 = 1.` and the fix will work without a crash. 

The muon GUI does not have the bug as it only uses the function browser.
Open muon analysis
Change the instrument to MUSR
Type 62260 into the run bar
Go to the fitting tab
Add a flat background
Right click on the parameter and select fit - no crash.

 @robertapplin should have some insights into this.

mantid-builder, 2023-01-26:
Fixed by #35081

closed 2023-01-26 by mantid-builder