mantidproject/mantid #34992: Manual Testing ORNL HFIR diffraction & 4Circle

sf1919, 2023-01-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

nan

Kvieta1990, 2023-02-08:
Done test for HB2A and HB2C for powder diffraction part and things are working as expected. Not sure about 4Circle.

Kvieta1990, 2023-02-08:
For powder part, I was testing out with `HB2AReduce` for HB2A and the following script for HB2C,

/HFIR/HB2C/shared/WANDscripts/HB2C_filter_parallel.py

Both are working as expected.

zjmorgan, 2023-02-10:
All looks good.

closed 2023-02-10 by peterfpeterson