mantidproject/mantid #34995: Manual Testing Step scan analysis

sf1919, 2023-01-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://www.mantidproject.org/Step_Scan_Interface

AndreiSavici, 2023-01-24:
The plot was no longer being normalized. Merged the fix

closed 2023-01-24 by AndreiSavici

AndreiSavici, 2023-01-24:
Tested against RHEL7 conda (mantidnightly at ORNL)

PR #35066 (2023-01-24): Normalize the data
Re #34995. Issue found that data is not normalized

**Description of work.**

**To test:**
Start StepScan interface. Load /SNS/HYS/IPTS-14344/nexus/HYS_111669.nxs.h5. Plot the data. Change normalization. See that the numbers on the y axis change.

Fixes #34995. 

*This does not require release notes* because **bug introduced in this release**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to  [... cut]