mantidproject/mantid #34998: Matplotlib deprecation warning when exporting cuts  in sliceviewer

RichardWaiteSTFC, 2023-01-17:
**Describe the bug**

**To Reproduce**
(1) Make workspace
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-3,3,-3,3,-3,3',
                   Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                   Frames='HKL,HKL,HKL',
                   SplitInto='2', SplitThreshold='10')
```
(2) Open sliceviewer
(3) Enable ROI tool and draw a rectangle
(4) Press `c` on keyboard

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

sf1919, 2023-04-11:
I am unable to recreate this bug on Windows 10 using Nightly version 6.6.20230407.0856. Maybe this has been fixed?

jhaigh0, 2023-06-23:
I also don't get a deprecation warning. @RichardWaiteSTFC Could you try to recreate this again?

RichardWaiteSTFC, 2023-06-23:
Yep I can't reproduce this anymore on IDAaaS using the smoke testing conda install so happy to close this if you are?

closed 2023-06-23 by jhaigh0