mantidproject/mantid #35017: Move the privacy policy from the archive page to the main docs.

DanielMurphy22, 2023-01-19:
The privacy policy, which is linked to on the About MantidWorkbench menu and on the error reporter, exists on the old archived webpages with screenshots from MantidPlot. This privacy policy should be moved to the main docs and have new screenshots.

_Originally posted by @DanielMurphy22 in https://github.com/mantidproject/mantid/issues/34945#issuecomment-1384073580_

martyngigg, 2023-01-19:
I have an overarching docs clean up I've been doing where I've included this so I'll pick this up.

DanielMurphy22, 2023-01-19:
Cool, tell us if you need any help with the docs cleanup!

martyngigg, 2023-01-19:
Thanks. A review will be helpful. I think I should have something to open tomorrow.

martyngigg, 2023-01-23:
I've opened #35049 that should clear up everything such that it's either current or points to archive.mantidproject.org if necessary.

PR #35049 (2023-01-30): Update all external links across codebase
**Description of work.**

This work assessed all links pointing to www.mantidproject.org. Those pointing to wiki pages that should have their documentation inside the code base have been moved to rst pages. Others have their addresses updated to point to archive.mantidproject.org

The MantidHelpWindow code has also been simplified so that a page can be called with a relative page path and it is assumed to be relative to the base paths of either the Qt Help or HTML documentation

**To test:**

To check the help window:

* Build the `docs-qthelp` target
 [... cut]
closed 2023-01-30 by DannyHindson