mantidproject/mantid #35018: UI elements appear small and inconsistent

MohamedAlmaki, 2023-01-19:
**Describe the bug**

When opening the nightly version on my machine some tab's UI elements appear small and inconsistent. This problem only appeared on my machine and not in other machines. Check the screenshot below to see the problem.
**To Reproduce**
1. Go to Interface->SANS->ISIS SANS
2. Go to the settings tab then Q, Wavelength, Detector Limits tab
3. You will see the UI is inconsistent and unusable

**Expected behavior**

The UI elements should appear clear and with reasonable size
**Screenshots**

[image]

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 10 Enterprise
 - Mantid Version 6.6.2023115.2237

**Additional context**

gemmaguest, 2023-01-19:
This is ok for me so is likely due to having a 4k display. This would be good to fix but I think this is unlikely to affect any of our users at the moment, especially as it's been a problem for a while, so I don't think it's urgent.

MohamedAlmaki, 2023-01-20:
> I see the same issue on Linux, on my 4k laptop monitor, if I use 200% scaling. With 100% scaling it looks right

Yes, for me the scaling is 250%. For me, with 100% it will look good but everything on the screen will be tiny. Also, notice that the problem is only present in this tab and not others in the same window.

closed 2024-01-23 by github-actions[bot]