mantidproject/mantid #35023: Unhandled Exception on Show/hide all errors for waterfall plot

DanielMurphy22, 2023-01-19:
Attempting to show or hide all errors on a waterfall plot will cause unhandled exception
[image]

```

Traceback (most recent call last):
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/workbench/plotting/figureerrorsmanager.py", line 120, in replot_curve
    new_curve = ax.replot_artist(curve, errorbars=curve_has_errors(curve), **plot_kwargs)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/plots/mantidaxes.py", line 509, in replot_artist
    self._remove_matching_curve_from_creation_args(workspace.name(), workspace_index, spec_num)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/plots/mantidaxes.py", line 1297, in _remove_matching_curve_from_creation_args
    raise ValueError("Curve does not have existing creation args")
ValueError: Curve does not have existing creation args

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/workbench/plotting/figureerrorsmanager.py", line 84, in update_plot_after
    func(*args, **kwargs)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/workbench/plotting/figureerrorsmanager.py", line 51, in toggle_error_bars_for
    new_curve = cls.replot_curve(ax, curve, plot_kwargs)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/workbench/plotting/figureerrorsmanager.py", line 122, in replot_curve
    new_curve = cls._replot_mpl_curve(ax, curve, plot_kwargs)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/workbench/plotting/figureerrorsmanager.py", line 137, in _replot_mpl_curve
    remove_curve_from_ax(curve)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/plotconfigdialog/curvestabwidget/__init__.py", line 95, in remove_curve_from_ax
    ax.remove_artists_if(lambda art: art == curve)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/plots/mantidaxes.py", line 337, in remove_artists_if
    return self._remove_workspace_artists_waterfall(predicate=unary_predicate)
  File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantid/plots/mantidaxes.py", line 352, in _remove_workspace_artists_waterfall
    if workspace.name() not in self.tracked_workspaces:
AttributeError: 'NoneType' object has no attribute 'name'

```

_Originally posted by @thomashampson in https://github.com/mantidproject/mantid/issues/34945#issuecomment-1384154820_

thomashampson, 2023-01-19:
This is a regression, so should be fixed for 6.6

mantid-builder, 2023-01-24:
Fixed by #35027

PR #35027 (2023-01-24): Fix error when adding/removing all error bars
**Description of work.**

Checks to see if the workspace is there before calling for its name. It is possible to call the function without a workspace and use a predicate instead, which meant this check was making calls on `None` objects. 

**To test:**
1. Load a workspace with multiple spectra.
2. Plot >1 in waterfall mode
3. Right click the plot and click Error Bars > Show All Errors
4. Error bars are shown and no exception is raised. 

Fixes #35023 

*This does not require release notes* because **it fixes a bug introduced in this release.**

-- [... cut]
closed 2023-01-24 by mantid-builder