mantidproject/mantid #35032: Update Mantid Basic Course

rbauststfc, 2023-01-19:
The following updates to the [Mantid Basic Course](http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course) were found to be needed during manual testing under issue #34955:

- [ ] [04_displaying_2D_data](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data) - does not have the correct image or any information relating to the cutting tool added in PR #33765.

- [ ] [06_formatting_plots.html](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html#formatting-plots) - plot toolbar graphic needs updating, it does not include the new hide button introduced in PR #34614.

- [ ] [1DPlotsHelp.html#basic-1d-plots](https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#basic-1d-plots) - all the script examples on this page are out of date (running them would give error "Legend has no attribute draggable"). It would be good to generate up to date examples from Mantid.

- [ ] [03_workspaces.html#correct-binning](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/03_workspaces.html#correct-binning) - plot generated from following the example gives [this colourfill](https://github.com/mantidproject/mantid/issues/34955#issuecomment-1385553972). Either need to update the screenshot in the course or correct the steps to allow users to produce a matching result.

- [ ] [03_investigating_data.html#the-toolbar](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#the-toolbar) - buttons on this page need updating.

- [ ] [03_investigating_data.html#selecting-arbitrary-sets-of-detectors](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#selecting-arbitrary-sets-of-detectors) - button positions have changed so the reference to positions here could do with being updated.

- [ ] [03_investigating_data.html#picking-a-single-detector](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#picking-a-single-detector) - following the steps exactly as written causes the plotted curve to disappear as soon as the mouse moves off the detector. We should make it clear that the necessary steps are to click the `Add a single crystal peak` button and then click a detector in the instrument view (not just mouse over one) before moving to zoom in to the mini-plot.

- [ ] [03_investigating_data.html#picking-a-tube](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#picking-a-tube) - similar to the above, except the necessary steps seem to be to click the tube selection button, click on a detector and then keep holding the mouse button down until the cursor is over the mini-plot. That way the plot's right click menu can then be used without the curve disappearing.

- [ ] [04_masking_and_grouping.html#region-of-interest](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/04_masking_and_grouping.html#region-of-interest) - the last paragraph of this section looks like it should actually be under the Masking section above.

- [ ] [06_sample_logs.html#basic-inspection](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_sample_logs.html#basic-inspection) - screenshot is of `BSS_11841_event` rather than `CNCS_7860_event`.

- [ ] [06_sample_logs.html#algorithms](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_sample_logs.html#algorithms) - the paragraph underneath the screenshot in this section that begins:
> Look at the effect this filter has on the Temperature Sample Log and...

should say SampleTemp rather then Temperature Sample. The explanation for the 0 relative time start is also confusing. Finally, the plot that should appear under this paragraph does not appear, it just incorrectly displays `(Source Code)`.

- [ ] [07_invalid_sample_logs.html](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/07_invalid_sample_logs.html) - the dataset `ENGINX00228061_log_alarm_data.nxs` referenced on this page should be added to the TrainingCourseData.

- [ ] [02_complex_case.html#functions-properties-setting-up-a-fit-model](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_complex_case.html#functions-properties-setting-up-a-fit-model) - the following paragraph is confusing as it mentions dragging on the plot after having set a FlatBackground, but this info should really be for the peak lines:
> Note that two fit functions have appeared in the Fit Property Browser called f0-Gaussian and f1-FlatBackground. Click on the triangles beside these functions to reveal their fit parameters. Just as with the StartX/EndX values, dragging on the plot and changing these values update each other.

- [ ] [02_complex_case.html#tying-and-constraining-fit-parameters](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_complex_case.html#tying-and-constraining-fit-parameters) - the `Tie > to Function` option works fine in the case of 2 peaks rather than 1 peak plus 1 background. The docs should be changed to make it clearer that the user should use `Tie > Custom Tie` for the examples it gives.

rbauststfc, 2023-02-10:
@DanielMurphy22, I think this item on the list of updates was referring to some data that you have:

[07_invalid_sample_logs.html](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/07_invalid_sample_logs.html) - the dataset ENGINX00228061_log_alarm_data.nxs referenced on this page should be added to the TrainingCourseData.

I'm not sure of the process for adding data to the training course, do you know how this works at all?

DanielMurphy22, 2023-02-13:
I don't think I can directly do it. I think you need to download the current TrainingCourseData from our Downloads page and unzip it. Then add in the new file, rezip and it needs to be uploaded to the same place on sourceforge. I'm not sure I have access to upload this, I think I had to ask someone else in the team to upload it for me: https://sourceforge.net/projects/mantid/files/Sample%20Data/TrainingCourseData.zip/download

rbauststfc, 2023-02-13:
Great, thanks, that's all really useful to know. I'm happy to sort out the new zip and arrange for it to be uploaded (plus check about adding all the sub-team leads to the admin list) if you can send me the data (or let me know where I can find it)?

DanielMurphy22, 2023-02-13:
Great, thanks! I'll send it on Slack!

DanielMurphy22, 2023-02-14:
For the second bullet point, the toolbar images are held in a powerpoint file here: https://github.com/mantidproject/documents/tree/main/Images/Images_for_Docs
I have just updated that document when needed and screenshotted the new toolbar to get a .png that looks good. As far as I remember the last time I had to take 2 screenshots and merge them as the toolbar has gotten quite long. Tell me if you'd rather I did this!

rbauststfc, 2023-02-14:
If you're happy to generate the new image @DanielMurphy22 then that would be amazing, thank you!

PR #35213 (2023-02-24): Update Mantid basic course
**Description of work.**

This PR makes the updates described on issue #35032 that were identified during manual testing for the 6.6 release. Note that point two (toolbar image updates) on the list of updates has been done in separate PR #35226.

**To test:**

1) Check that the text changes read OK.
1) Build the docs-html target to check that the documentation builds and that the new screenshots render OK.

Fixes #35032. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

### [... cut]
closed 2023-02-24 by rosswhitfield