mantidproject/mantid #35048: Fake Live Data crash for incorrect path

DanielMurphy22, 2023-01-20:
https://github.com/mantidproject/mantid/issues/34954#issuecomment-1385697957

Found during [Manual Testing](https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html) so may not be affecting users.

**Describe the bug**
When the path to the fileeventdatalistener.filename property is wrong, and you try to select the ADARA_FileReader instrument in Load LiveData, MantidWorkbench crashes and closes

**To Reproduce**
Follow the live data testing section ADARA Fake Event
 and set the following lines in your Mantid.user.properties file.
```
fileeventdatalistener.filename=REF_L_32035_neutron_event.dat
fileeventdatalistener.chunks=300
```
If the file path is set incorrectly, such as just the filename or with "/home/username/Desktop/Data/REF_L_32035_neutron_event.dat" (rather than the correct "/Users/username/Desktop/Data/REF_L_32035_neutron_event.dat")...
When you try to Load > Live Data MantidWorkbench will crash and close
**Expected behavior**
It does not crash and offers a useful error message.

**Platform/Version (please complete the following information):**
 - OS: macOS Monterey
 - Mantid Version 6.6 manual testing nightly 15th Jan

jhaigh0, 2023-02-16:
When I run the `LoadLiveData` algorithm I don't see ADARA_FileReader in the instrument list.
[image]

Do I need to change to a different facility?

DanielMurphy22, 2023-02-16:
Change your facility to `TEST_LIVE`. Sorry the instructions aren't great, it mentions this in the manual testing instructions.

jhaigh0, 2023-09-05:
Still should be solved

closed 2024-03-11 by github-actions[bot]