mantidproject/mantid #35073: Fix failing doc-tests

peterfpeterson, 2023-01-24:


gecage952, 2023-01-24:
From doing a full run locally the following doctests are failing for me (someone should verify):
- [ ] algorithms/ConvertSnsRoiFileToMask-v1.rst - test and cleanup code
- [ ] algorithms/ConvolveWorkspaces-v1.rst
- [ ] algorithms/CorelliCalibrationDatabase-v1.rst
- [ ] algorithms/CorrectTOFAxis-v1.rst - 2 failures test
- [ ] algorithms/CreateEPP-v1.rst
- [ ] algorithms/CreateSampleWorkspace-v1.rst
- [ ] algorithms/EnggCalibrateFull-v1.rst - test and cleanup code
- [ ] algorithms/FindEPP-v2.rst
- [ ] algorithms/FindPeaks-v1.rst
- [ ] algorithms/Fit-v1.rst
- [ ] algorithms/FitGaussian-v1.rst
- [ ] algorithms/FitPeaks-v1.rst
- [ ] algorithms/GetDetectorOffsets-v1.rst
- [ ] algorithms/HB3AIntegrateDetectorPeaks-v1.rst
- [ ] algorithms/IntegrateEPP-v1.rst
- [ ] algorithms/Integration-v1.rst
- [ ] algorithms/LoadDetectorsGroupingFile-v1.rst- 2 failures test - 2 cleanup code failures
- [ ] algorithms/LoadEMU-v1.rst
- [ ] algorithms/LoadILLTOF-v2.rst
- [ ] algorithms/LoadNXcanSAS-v1.rst
- [ ] algorithms/LoadSESANS-v1.rst- Test and cleanup code
- [ ] algorithms/MatchPeaks-v1.rst
- [ ] algorithms/MayersSampleCorrection-v1.rst
- [ ] algorithms/MSDFit-v1.rst
- [ ] algorithms/SaveDaveGrp-v1.rst
- [ ] algorithms/SaveFITS-v1.rst- just cleanup code
- [ ] algorithms/SaveMD-v1.rst
- [ ] algorithms/SaveMD-v2.rst - test and cleanup code
- [ ] algorithms/SaveNXcanSAS-v1.rst
- [ ] algorithms/SaveVulcanGSS-v1.rst - test and cleanup code
- [ ] algorithms/SpatialGrouping-v1.rst- Test and cleanup code
- [ ] algorithms/SpecularReflectionPositionCorrect-v2.rst
- [ ] techniques/PolarisedDiffractionILL.rst

AnthonyLim23, 2023-01-31:
algorithms/LoadEMU-v1.rst is doing the wrong thing. It should not include monitors https://github.com/mantidproject/mantid/issues/35118

peterfpeterson, 2023-01-31:
@AnthonyLim23 EMU is an instrument at ANSTO

AnthonyLim23, 2023-01-31:
@geishm-ansto should LoadEMU include monitors?

geishm-ansto, 2023-01-31:
@AnthonyLim23 EMU did not include a beam monitor. It is a backscattering instrument with a doppler shift and the scientist relied on the direct measurements as a pseudo-monitor for flux normalisation.

AnthonyLim23, 2023-02-01:
@geishm-ansto so does the algorithm need changing to not include the monitors?

ConorMFinn, 2023-02-01:
@geishm-ansto The instrument definition includes this extra bank, which are loaded in as monitors. Is this new/expected? 
[image]

geishm-ansto, 2023-02-01:
@ConorMFinn The history with the EMU instrument is that the (pseudo beam monitor) tube was always there but without electronics connected so there was no data and the original IDF did not include the spare tube. Now that it is active the IDF was updated. The early data files could be loaded with the new IDF but have zero events recorded for that tube. When the change was submitted the doc-tests passed at the time (?). Your change sounds fine, thanks.

PR #35133 (2023-02-23): Fix failing doctests
**Description of work.**

Updated doctests to take into account other code changes. 

**To test:**
1. Build workbench and `AllTests` target. Then open workbench and in ManangeUserDirectories add the paths to the UnitTests and DocTests in the test data. Close workbench.
2. Build the `docs-doctest` target and check all tests pass

Fixes #35073 

**Note: One of the jenkins pipelines will need updating to add the `--enable-doctests` to the buildscript call. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/Reviewi [... cut]
closed 2023-02-23 by peterfpeterson