mantidproject/mantid #35097: ALFView Normalisation error when X arrays do not match after convertunits

robertapplin, 2023-01-27:
**Original reporter:** [username facility]/[nobody]
Russell Ewings

**Describe the bug**
A bug preventing the normalisation of ALF data on the ALFView interface has been found. The error message “Vanadium normalisation failed: X arrays must match when dividing two workspaces” appears, and so they can't continue with the rest of their workflow.

**To Reproduce**
Russells email:
```
I’ve run in to a problem with ALFView when testing the nightly version. I get an error message when attempting the vanadium normalisation, saying “Vanadium normalisation failed: X arrays must match when dividing two workspaces”. If I load the two datasets into Mantid manually and, with no additional processing, divide one by the other there is no problem.

If you want to see if you can reproduce this, then the sample run is #84116, which can be found in [\\isis\inst$\NDXALF\Instrument\data\cycle_20_3](file://isis/inst$/NDXALF/Instrument/data/cycle_20_3) and the vanadium run is #78634, which is in [\\isis\inst$\NDXALF\Instrument\data\cycle_18_2](file://isis/inst$/NDXALF/Instrument/data/cycle_18_2)
```

**Expected behavior**
It should be possible to use this vanadium file to normalise the sample file. To fix this, we need to use the `RebinToWorkspace` algorithm to rebin the vanadium file based on the sample file after the conversion to dSpacing, which can yield differing bins.

**Platform/Version (please complete the following information):**
 - OS: All
 - Mantid nightly

robertapplin, 2023-01-30:
It seems this is not quite fixed

mantid-builder, 2023-01-30:
Fixed by #35101

PR #35117 (2023-02-01): Normalise sample by vanadium before dSpacing conversion
**Description of work.**
This PR ensures the normalisation of the sample by the vanadium is performed before converting the result to dSpacing. If you convert the sample and vanadium to dSpacing first, the resulting normalised data is incorrect.

The problem was spotted by Russell during beta testing.

**To test:**
Open ALFView
Load ALF84116 as sample
Load ALF78634 as vanadium
There should be no error message, and a plot should appear in the instrument widget:
[image]

Fixes #35097

*This does not require release notes* because **vanadium [... cut]
closed 2023-01-30 by mantid-builder