mantidproject/mantid #35134: Engineering Diffraction interaction with GSAS-II not working on Windows

DannyHindson, 2023-02-02:
The GSAS tab on the Engineering Diffraction UI provides a call from Mantid to GSAS-II. This has stopped working on Windows recently and fails with the following error. The message is quite long but the main problem seems to be that numpy isn't able to load the mkl-service python package (Intel math kernel library):
```
GSAS-II call must have failed, as the output project file was not found.

---------------------
---------------------
 Errors from GSAS-II 
---------------------
---------------------
C:\Users\iao46725\gsas2full\lib\site-packages\numpy\__init__.py:138: UserWarning: mkl-service package failed to import, therefore Intel(R) MKL initialization ensuring its correct out-of-the box operation under condition when Gnu OpenMP had already been loaded by Python process is not assured. Please install mkl-service package, see http://github.com/IntelPython/mkl-service
  from . import _distributor_init
Traceback (most recent call last):
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\__init__.py", line 22, in <module>
    from . import multiarray
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\multiarray.py", line 12, in <module>
    from . import overrides
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: DLL load failed: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\mantid\qt\python\mantidqtinterfaces\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\gsas2\call_G2sc.py", line 194, in <module>
    import GSASIIscriptable as G2sc  # noqa: E402
  File "C:\Users\iao46725\gsas2full\GSASII\GSASIIscriptable.py", line 1266, in <module>
    import numpy.ma as ma
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\__init__.py", line 140, in <module>
    from . import core
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "C:\Users\iao46725\gsas2full\python.exe"
  * The NumPy version is: "1.19.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: DLL load failed: The specified module could not be found.

---------------------
---------------------

GSAS-II call must have failed, as the output result file was not found.

---------------------
---------------------
 Errors from GSAS-II 
---------------------
---------------------
C:\Users\iao46725\gsas2full\lib\site-packages\numpy\__init__.py:138: UserWarning: mkl-service package failed to import, therefore Intel(R) MKL initialization ensuring its correct out-of-the box operation under condition when Gnu OpenMP had already been loaded by Python process is not assured. Please install mkl-service package, see http://github.com/IntelPython/mkl-service
  from . import _distributor_init
Traceback (most recent call last):
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\__init__.py", line 22, in <module>
    from . import multiarray
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\multiarray.py", line 12, in <module>
    from . import overrides
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: DLL load failed: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\mantid\qt\python\mantidqtinterfaces\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\gsas2\call_G2sc.py", line 194, in <module>
    import GSASIIscriptable as G2sc  # noqa: E402
  File "C:\Users\iao46725\gsas2full\GSASII\GSASIIscriptable.py", line 1266, in <module>
    import numpy.ma as ma
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\__init__.py", line 140, in <module>
    from . import core
  File "C:\Users\iao46725\gsas2full\lib\site-packages\numpy\core\__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "C:\Users\iao46725\gsas2full\python.exe"
  * The NumPy version is: "1.19.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: DLL load failed: The specified module could not be found.

---------------------
---------------------
```

DannyHindson, 2023-02-02:
I've gone back and checked when the problem started and it happened when this PR was merged: https://github.com/mantidproject/mantid/pull/34911

This seems to have caused the version of the (Windows only) mkl conda package to update from 2020.4 to 2022.1.0 and this has stopped GSAS-II being able to run - at least when run from Mantid. It seems to start up when run standalone. I can fix the problem by pinning mkl to 2020.4 but this is a bit of a bodge. GSAS-II shouldn't care or know about what conda packages Mantid has so there's some unwanted cross-talk here.

DannyHindson, 2023-02-02:
I think the problem is caused by the fact we're not activating the GSAS2 conda environment before calling G2sc. Will add a fix but will probably mimic activating the conda environment by editing the PATH environment variable. Calling the conda activation script before the python g2sc library has some complexity around transferring the PATH environment variable updates from one step to the next + bit too much heavy lifting close to a release

mantid-builder, 2023-02-03:
Fixed by #35139

closed 2023-02-03 by mantid-builder