mantidproject/mantid #35194: Upgrade OpenSSL to 3.0.8

sf1919, 2023-02-09:
**Describe the outcome that is desired.**
All versions of OpenSSL v1.1.1 are being deprecated in September 2023. We need to move to a newer version of OpenSSL that is supported - v3.0.8 potentially.

**Additional context**
This may need considerable work to make code base compatible with upgraded version of OpenSSL

MialLewis, 2023-02-13:
To note, it looks as if it's our `qt` pin which is preventing us from upgrading to `3.0.8`.

sf1919, 2023-02-14:
Once this work has been completed the version of OpenSSL that we use in our Linux docker containers will also need to be upgraded - https://github.com/mantidproject/dockerfiles/issues/70

sf1919, 2023-04-11:
Full details of EOL - https://www.openssl.org/blog/blog/2023/03/28/1.1.1-EOL/

PR #35434 (2023-04-26): Upgrade Qt to 5.15.8
**Description of work.**
Upgrade Qt version from 5.15.4 to 5.15.8. It was previously pinned to 5.15.4 to avoid GUI rendering problems, but these seem to have been resolved now. This is required before #34820 can be addressed.
Moving to 5.15.8 has meant that #35029 has become a problem again.

Other changes:
- Old Poco pin was holding back Qt (I think), so this has been updated. It has been restricted to avoid v1.12.2 to 1.12.4 because they added an unguarded `#define NOMINMAX` that introduced a compiler warning because we define it ourselves in WindowsSetup.cmake. They've  [... cut]
closed 2023-04-26 by tweber-ill