mantidproject/mantid #35214: Issue displaying plot in Mantid Basic Course

rbauststfc, 2023-02-15:
The plot underneath this paragraph doesn't appear in the published version of the Mantid basic course: https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_sample_logs.html#algorithms 

The page just  incorrectly displays (Source Code). However, when the docs are built locally (with the `DOCS_PLOTDIRECTIVE` CMake option turned on) the plot does seem to appear:

[image]

I've only tested this on a local Windows build. We should investigate why this plot isn't being generated in the published version of the docs.

Note that the explanation for the 0 relative time start in the paragraph above the plot is confusing, so this should be clarified as part of updating this if possible.

jhaigh0, 2023-08-21:
When I've tested building the docs on both Windows and Ubuntu, the plot appears as it should do. Very strange one.

rbauststfc, 2023-08-21:
Thanks for taking a look @jhaigh0, have you been building locally? I think I found the local build was OK, but for some reason the plot doesn't appear when published, so maybe an issue somewhere in the Jenkins build/publishing pipeline (in which case this might be one for Dev Ops)?

jhaigh0, 2023-08-21:
Yes just locally on my laptop and then on a VM. I think you might be right

jclarkeSTFC, 2023-11-14:
It looks like the published docs now have the plots, should we close this?

rbauststfc, 2023-11-14:
Great, yes, we can close this now that it seems to be working OK.

closed 2023-11-14 by jclarkeSTFC