mantidproject/mantid #35224: Provide a post-fitting check in PDCalibration/FitPeaks for I/Sigma ratio.

RichardWaiteSTFC, 2023-02-16:
**Describe the solution you'd like**
Be able to provide minimum I/sigma  in `PDCalibration`  (and possibly `FitPeaks`) for inclusion of peak centres in fit to determine diffractometer constants.

**Additional context**
Came up during discussion of PR #35094.
@bolotovskyr has opened this issue as a story in the Engineering Workflow Management system: `Story 356 [Enabler] Provide a post-fitting check in PDCalibration/FitPeaks for I/Sigma ratio`

bolotovskyr, 2023-03-07:
I think the implementation would involve the following tasks:

-     Add a new property to the `PDCalibration` class, e.g."MinimumPeakIntensityToSigmaRatio"
-     After FitPeaks is executed, calculate I/Sigma for every fitted peak: 
The integrated intensity `I` can be calculated as a sum over all data points in the peak window of the corresponding `(peak function - background function)` values.  The estimated error of the integrated intensity `Sigma` can be calculated by propagating the errors of all fitted parameters used in the `I` calculation. This would require calculating partial derivatives of the peak and background functions with respect to their parameters.   Looking at the source code, `PDCalibration` may use the following peak functions: `BackToBackExponential`, `Gaussian`, `Lorentzian`, `PseudoVoigt`, `IkedaCarpenterPV`  and the following background functions: `Flat`, `Linear`, `Quadratic`.
-     Reject fitted peaks that do not satisfy the minimum I/Sigma ratio.
Note, `PDCalibration` already has a mechanism for rejecting "bad" peaks, for example based on the fitted peak height or chi-square.

RichardWaiteSTFC, 2023-03-07:
Thanks for the break-down!

> The estimated error of the integrated intensity Sigma can be calculated by propagating the errors of all fitted parameters used in the I calculation. This would require calculating partial [...]

Due to the number of different functions I propose we get the error on the integrated intensity from a quadrature sum of the errors on the individual points over a region for which the peak function (i.e. fit curve - background) is some threshold e.g. 1% of the height as is done here (although not assuming raw counts - i.e. the error in each bin may not be e=sqrt(y) - for example if the data have been normalised).

https://docs.mantidproject.org/nightly/algorithms/IntegratePeaksProfileFitting-v1.html#integrating-the-model

Note that the situation is slightly complicated by the fact some fit function have the height as the fit parameter and other such as the `BackToBackExpoential` and `IkedaCarpenter` have the area (though for those functions I suppose we could get the error from the covariance matrix - but I think it's better to treat all peaks equally)

RichardWaiteSTFC, 2023-04-20:
@mguthriem also suggested a check on width (larger than 1 bin width)  - could also say width shouldn't be larger than the window to rule out the peak fitting broad background as well.

PR #37826 (2024-09-12): Add signal to sigma post fitting check
### Description of work
This pull request introduces a new property to FitPeaks and PDCalibration: the option to set a minimum signal-to-sigma ratio for post-fitting. When this property is provided, peaks with a signal-to-sigma ratio below the specified minimum will be rejected. The signal-to-sigma ratio is calculated as the integrated fitted peak minus the integrated background, divided by the sigma, which is calculated using the quadrature sum of the errors.

Fixes #35224. 

### To test:
1- Run the following script:
```
from mantid.simpleapi import *

w [... cut]
closed 2024-09-12 by github-actions[bot]